RoboStack/ros-jazzy #3: Request to Assist as Contributor/Maintainer

oursland, 2024-09-08:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I'd like to assist in getting the `ros-jazzy` RoboStack distribution built and operational.

I have been using `ros-humble` but newer packages are no longer building with the older LTS distribution.  Of note, I am aiming to build and test `rmw_zenoh` to address substantial shortcomings in the DDS implementations on macOS.

### Installed packages

```shell
---
```

### Environment info

```shell
---
```

Tobias-Fischer, 2024-09-14:
Hi @oursland - sorry I’ve been flat out with paper/grant submissions and teaching. We’re always very happy for contributions! I see you opened a PR in #4 - let’s see how it goes. I typically recommend building locally first; unfortunately sometimes CI has some additional issues though but I can try to take a look if that happens.

oursland, 2024-09-15:
Hi @Tobias-Fischer !  I understand that struggle; I'm in the midst of it myself (minus teaching).

I have been fairly active with my progress on the RoboStack/Lobby channel on Gitter.  At this point I can build most of `ros-jazzy-desktop` with the exception of the Python Qt bindings and their dependencies (`rqt` being a major one) and a couple other items.  I'll get to working on the Qt bindings issue very soon as my current project depends upon `rqt` being functional.

The CI has some issues that will require that the `rattler-build` branch of `vinca` being updated to address.  I have submitted https://github.com/RoboStack/vinca/pull/60 and https://github.com/RoboStack/vinca/pull/58 to address these issues.  Once merged I'll look at correcting the CI.

Tobias-Fischer, 2024-09-15:
Sounds great, thank you! The qt bindings have always been a pain ..

I’ve merged the two vinca PRs.

Tobi

oursland, 2024-09-15:
This PR should fix CI for `osx` and `osx_arm64`: https://github.com/RoboStack/ros-jazzy/pull/5

This PR should then permit building of `ros-jazzy-desktop` for `osx` and `osx_arm64`: https://github.com/RoboStack/ros-jazzy/pull/4

Tobias-Fischer, 2025-01-21:
Hi @oursland - did you ever try zenoh? Would love to hear how well it works. I've got an interest in removing our custom RMW for ROS2WASM (https://www.arxiv.org/abs/2409.09941) as zenoh should support WebAssembly.

/cc @nmarticorena @IsabelParedes

oursland, 2025-01-21:
Thus far I have only performed some basic ROS 2 tests to confirm functionality with the RoboStack `ros-jazzy`, which it passed.  In that regard, it worked quite well.

Next week I intend on conducting initial experiments with mesh networking in a Kubernetes environment for an upcoming proposal.  This is something that DDS does not do well as most Kubernetes networking layers do not support UDP multicast or do so poorly.  `rmw_zenoh` should be able to get around this limitation handily through multiple means including the use of a Zenoh Router for discovery.  I will provide more feedback after I have conducted these experiments.

traversaro, 2025-01-22:
By the way, I think we never build `zenoh_cpp_vendor` . FYI, this is a package that we may want to devendor using https://github.com/conda-forge/zenoh-cpp-feedstock .

traversaro, 2025-01-22:
> By the way, I think we never build `zenoh_cpp_vendor` . FYI, this is a package that we may want to devendor using https://github.com/conda-forge/zenoh-cpp-feedstock .

Related humble PR: https://github.com/RoboStack/ros-humble/pull/253 , we can do something similar also for jazzy.

Tobias-Fischer, 2025-04-14:
Let's close here - there have been quite a few nice community contributions which is awesome, and always very welcome :)!

closed 2025-04-14 by Tobias-Fischer