RoboStack/ros-jazzy #26: RViz2 Bus Error

ffugler99, 2025-02-08:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

After following the instructions in the Getting Started guide to install ros-jazzy-desktop, I attempt to test the installation (within the activated mamba environment) by running rviz2. The GUI opens for a few seconds, then crashes with this message in the terminal:

[INFO] [1738982352.087628000] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1738982352.087665000] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[INFO] [1738982352.120429000] [rviz2]: Stereo is NOT SUPPORTED
2025-02-07 19:39:12.185 rviz2[5696:38527] +[IMKClient subclass]: chose IMKClient_Modern
2025-02-07 19:39:12.185 rviz2[5696:38527] +[IMKInputSession subclass]: chose IMKInputSession_Modern
zsh: bus error  rviz2

### Installed packages

```shell
# packages in environment at /opt/homebrew/Caskroom/miniforge/base/envs/ros_env:
#
# Name                    Version                   Build  Channel
aiohappyeyeballs          2.4.4              pyhd8ed1ab_1    conda-forge
aiohttp                   3.11.12         py311h4921393_0    conda-forge
aiosignal                 1.3.2              pyhd8ed1ab_0    conda-forge
aom                       3.9.1                h7bae524_0    conda-forge
argcomplete               3.5.2              pyhd8ed1ab_0    conda-forge
assimp                    5.4.3                ha9c0b8d_0    conda-forge
atk-1.0                   2.38.0               hd03087b_2    conda-forge
attrs                     25.1.0             pyh71513ae_0    conda-forge
blosc                     1.21.6               h7dd00d9_1    conda-forge
brotli                    1.1.0                hd74edd7_2    conda-forge
brotli-bin                1.1.0                hd74edd7_2    conda-forge
bullet                    3.25                 hf3b2ce4_3    conda-forge
bullet-cpp                3.25            py311h57524c7_3    conda-forge
bzip2                     1.0.8                h99b78c6_7    conda-forge
c-ares                    1.34.4               h5505292_0    conda-forge
c-compiler                1.9.0                hdf49b6b_0    conda-forge
ca-certificates           2025.1.31            hf0a4a13_0    conda-forge
cairo                     1.18.2               h6a3b0d2_1    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_1    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_1    conda-forge
cctools                   1010.6               hb4fb6a3_3    conda-forge
cctools_osx-arm64         1010.6               h3b4f5d3_3    conda-forge
cffi                      1.17.1          py311h3a79f62_0    conda-forge
clang                     18.1.8          default_h474c9e2_7    conda-forge
clang-18                  18.1.8          default_hf90f093_7    conda-forge
clang_impl_osx-arm64      18.1.8              h2ae9ea5_23    conda-forge
clang_osx-arm64           18.1.8              h07b0088_23    conda-forge
clangxx                   18.1.8          default_h1ffe849_7    conda-forge
clangxx_impl_osx-arm64    18.1.8              h555f467_23    conda-forge
clangxx_osx-arm64         18.1.8              h07b0088_23    conda-forge
cli11                     2.4.2                h286801f_0    conda-forge
cmake                     3.31.5               ha25475f_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_1    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_1    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_1    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_2    conda-forge
colcon-core               0.18.2             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.3.0              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.4.0              pyhd8ed1ab_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.3.0              pyhd8ed1ab_1    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
colcon-python-setup-py    0.2.9              pyhff2d567_0    conda-forge
[... 617 lines cut ...]
         python version : 3.12.8.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=m1
                          __conda=24.11.3=0
                          __osx=15.3=0
                          __unix=0=0
       base environment : /opt/homebrew/Caskroom/miniforge/base  (writable)
      conda av data dir : /opt/homebrew/Caskroom/miniforge/base/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-jazzy/osx-arm64
                          https://conda.anaconda.org/robostack-jazzy/noarch
                          https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /opt/homebrew/Caskroom/miniforge/base/pkgs
                          /Users/fisherfugler/.conda/pkgs
       envs directories : /opt/homebrew/Caskroom/miniforge/base/envs
                          /Users/fisherfugler/.conda/envs
               platform : osx-arm64
             user-agent : conda/24.11.3 requests/2.32.3 CPython/3.12.8 Darwin/24.3.0 OSX/15.3 solver/libmamba conda-libmamba-solver/24.9.0 libmambapy/1.5.12
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

traversaro, 2025-02-08:
Thanks for reporting! Just to double check, do you get the same error if you run rviz2 in bash?  Can you report the exact command you used to run rviz2?

ffugler99, 2025-02-09:
@traversaro I'm using
```
mamba activate ros_env
rviz2
```

I just tried with bash, rviz2 ran for a couple minutes seemingly with no issues, but then crashed with another bus error:
```
[INFO] [1739060485.199354000] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1739060485.199391000] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[INFO] [1739060485.255974000] [rviz2]: Stereo is NOT SUPPORTED
2025-02-08 17:21:25.800 rviz2[13104:212163] +[IMKClient subclass]: chose IMKClient_Modern
2025-02-08 17:21:25.800 rviz2[13104:212163] +[IMKInputSession subclass]: chose IMKInputSession_Modern
Bus error: 10
```

traversaro, 2025-02-09:
Can you try launching rviz2 with `ros2 run rviz2 rviz2`, as suggested in rviz documentation?  https://docs.ros.org/en/humble/Tutorials/Intermediate/RViz/RViz-User-Guide/RViz-User-Guide.html#startup

ffugler99, 2025-02-12:
Same issue there:
```
(ros_env) bash-3.2$ ros2 run rviz2 rviz2
[INFO] [1739325476.938341000] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1739325476.938377000] [rviz2]: OpenGl version: 2.1 (GLSL 1.2)
[INFO] [1739325476.993223000] [rviz2]: Stereo is NOT SUPPORTED
2025-02-11 18:57:57.220 rviz2[31591:585667] +[IMKClient subclass]: chose IMKClient_Modern
2025-02-11 18:57:57.220 rviz2[31591:585667] +[IMKInputSession subclass]: chose IMKInputSession_Modern
[ros2run]: Bus error: 10
```

closed 2025-07-27 by Tobias-Fischer