RoboStack/ros-jazzy #37: teleop_twist_joy does not work on MacOSX

jfrancis71, 2025-03-06:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

teleop_twist_joy does not work on MacOSX:

``` ros2 launch teleop_twist_joy teleop-launch.py joy_config:='xbox'```

It starts up fine. But when I move the joystick it prints:
[joy_node-1] [INFO] [1741248056.857958000] [joy_node]: Unknown event type 1544

and does not generate a twist message.

I believe the error message originates from:
https://github.com/ros-drivers/joystick_drivers/blob/ros2/joy/src/joy.cpp
Line: 451

```} else {
        RCLCPP_INFO(get_logger(), "Unknown event type %d", e.type);
      }
```

I note this works fine on Linux.

If I do (on Linux):
```mamba list | grep "sdl"```
I get:
ros-jazzy-sdl2-vendor     3.3.0           np126py311h15c07e3_1    robostack-jaz
zy
sdl2                      2.30.10              h63c27ac_0    conda-forge

But when I do the same on my Mac I get:
  ros-jazzy-sdl2-vendor                             3.3.0         np126py311hd6
9bdec_4      robostack-jazzy
  sdl2                                              2.32.50       hc0cb955_1   
             conda-forge    
  sdl3                                              3.2.8         h6dd79e8_0   
             conda-forge

I am suspicious that there may be some mixing of sdl versions. There are no sdl
's in my base environment, and it seems it is the jazzy desktop mamba install w
hich is installing all 3 sdl's on my Mac. (As confirmed by fresh install in new
 environment).

Any help much appreciated.

Thanks,
Julian.

### Installed packages

```shell
github won't allow me to submit report with packages listed here.
```

### Environment info

```shell
libmamba version : 2.0.5
          mamba version : 2.0.5
           curl version : libcurl/8.12.1 OpenSSL/3.4.1 (SecureTransport) zlib/1.3.1 zstd/1.5.7 libssh2/1.11.1 nghttp2/1.64.0
     libarchive version : libarchive 3.7.7 zlib/1.2.13 liblzma/5.4.6 bz2lib/1.0.8 liblz4/1.9.4 libzstd/1.5.2
       envs directories : /Users/julian/miniconda3/envs
          package cache : /Users/julian/miniconda3/pkgs
                          /Users/julian/.mamba/pkgs
            environment : ros_a1 (active)
           env location : /Users/julian/miniconda3/envs/ros_a1
      user config files : /Users/julian/.mambarc
 populated config files : /Users/julian/miniconda3/envs/ros_a1/.condarc
                          /Users/julian/.condarc
       virtual packages : __unix=0=0
                          __osx=12.7.6=0
                          __archspec=1=x86_64_v3
               channels : https://conda.anaconda.org/robostack-jazzy/noarch
                          https://conda.anaconda.org/robostack-jazzy/osx-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/robostack-staging/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
       base environment : /Users/julian/miniconda3
               platform : osx-64
```

jfrancis71, 2025-03-06:
Just to clarify in my report above, the formatting of my mamba list | grep "sdl" output  was not correct (c/r in wrong place).

On my MacOS system (which fails):
  ros-jazzy-sdl2-vendor                             3.3.0            np126py311hd69bdec_4      robostack-jazzy
  sdl2                                                             2.32.50       hc0cb955_1                            conda-forge    
  sdl3                                                             3.2.8             h6dd79e8_0                           conda-forge    

whereas Linux (which works):
ros-jazzy-sdl2-vendor                              3.3.0              np126py311h15c07e3_1       robostack-jazzy
sdl2                                                              2.30.10          h63c27ac_0                          conda-forge

traversaro, 2025-03-06:
Can you try to install sdl2 2.30.0 also on macOS to understand if that fixes the issue?

jfrancis71, 2025-03-07:
I was not able to find an sdl2 version 2.30.0. My suspicion that there may be mixing of sdl2 and sdl3 is not correct.

I have done a fresh mamba environment install.

I run the teleop joy command  and this fails as detailed in opening comment.

I follow your suggestion (modified):
mamba install sdl2=2.30.10     # this matches the Linux version

I rerun teleop joy. The error message has gone. I am now successfully publishing Twist messages.

So this solves the problem. Let me know if I need to close the issue. I have not done so as I assume there may be a change required to the conda package to solve this for other Mac users.

Thank you for your help,
Julian.

jfrancis71, 2025-03-07:
Would you like me to respond to your comments here, or in your link above?

traversaro, 2025-03-07:
> Would you like me to respond to your comments here, or in your link above?

As you prefer. Probably as the discussion started here it make sense to continue here, that issue is just for visibility for people checking the sdl repo on conda-forge.

jfrancis71, 2025-03-07:
I am using XEOX Gamepad SL-6556-BK
This is a very old joystick (but still works...)

I have executed on my Ubuntu 22.04.5 LTS
```mamba install sdl2==2.32.50```

So now my teleop joy is no longer working on my Linux box (as expected).
It fails in exactly the way described before.

I have just restored 2.30.10, and it is back working again.

traversaro, 2025-03-07:
Perfect, thanks! Let me edit the title of the issue for clarity.

traversaro, 2025-03-07:
TL;DR: for anyone reading this issue: a working workaround is to add a constraint for `sdl2 <= 2.30.*` to get it working again.

jfrancis71, 2025-04-02:
Just wondering if there is likely to be an update on this? I'd quite like to move all my repos to Jazzy, but I'm a bit reluctant to do so as I have to add in extra instructions for Mac users to get around this bug.

My setup is my robots run on ARM, but it is super convenient to control them with teleop-joy running on Macbook (because it is relatively portable).

traversaro, 2025-04-02:
Hell0 @jfrancis71, unfortunately I do not have any joystick, so it is tricky for me to debug this. Something that may help would be to reproduce this issue without ROS, using the SDL2 package to read the joystick and check if it does not work.

jfrancis71, 2025-04-02:
Thanks for your reply. I have managed to build the ROS2 joy package directly and run it. I actually think it is not an error in ROS2 (as far as I can see). It's probably fairly easy to strip out all the ROS2 stuff for an example, but I have not thought of a super simple way of generating the actual joystick events without a physical joystick, which I think you would need to test it properly.

I have noticed another possibly related thread to which I have commented: https://github.com/libsdl-org/sdl2-compat/issues/424
Note ROS2 event message 1544 is in decimal, but this is 0x608 in hexadecimal, hence my interest in the above thread.

traversaro, 2025-04-02:
> Looks like this will be resolved with [conda-forge/sdl-feedstock#28](https://github.com/conda-forge/sdl-feedstock/pull/28)

Great, this turned out to be easier then expected (also as we do not need to fix anything on the robostack side, the newer version of sdl2-compat will be used as soon as it is available.

jfrancis71, 2025-04-02:
Super. I look forward to this. I'll monitor this post (albeit it may be tomorrow).

jfrancis71, 2025-04-04:
On my Linux computer I have installed sdl2=2.32.54 and rerun teleop=joy. No error messages and twist topic broadcasted correctly.

On my MacBook I have created a fresh RoboStack Jazzy install as per Robostack instructions.It installs sdl2=2.32.54 by default.
I run teleop joy. Again no error messages and twist topic broadcasting correctly.

I believe the problem is solved. Thank you very much.

closed 2025-04-04 by jfrancis71

traversaro, 2025-04-04:
Great, thanks a lot @jfrancis71 !