RoboStack/ros-jazzy #43: Build on main branch broken [unix]

Tobias-Fischer, 2025-04-14:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

The unix builds fail as the packages are uploaded to `robostack-jazzy`, but then we look for them in `robostack-staging`.

@traversaro - was the intended behaviour that we still upload to `robostack-staging` and then copy over to `robostack-jazzy`?

### Installed packages

```shell
n/a
```

### Environment info

```shell
n/a
```

traversaro, 2025-04-14:
>  @traversaro - was the intended behaviour that we still upload to robostack-staging and then copy over to robostack-jazzy?

robostack-jazzy packages were never uploaded to robostack-staging, were always directly uploaded to robostack-jazzy . Perhaps there was a regression in vinca?

Tobias-Fischer, 2025-04-14:
Fixed with https://github.com/RoboStack/ros-jazzy/commit/713c937dfb5963335d5055e9c8e3a607be75c674 - thanks for the hint!

traversaro, 2025-04-15:
> Fixed with https://github.com/RoboStack/ros-jazzy/commit/713c937dfb5963335d5055e9c8e3a607be75c674 - thanks for the hint!

I see, copy paste error!

traversaro, 2025-04-16:
Let's reopen as there are other failures:

~~~
 │ │ -- Found Ceres version: 2.2.0 installed in: $PREFIX with components: [EigenSparse, SparseLinearAlgebraLibrary, LAPACK, SuiteSparse, SchurSpecializations]
 │ │ -- Found cartographer_ros_msgs: 2.0.9003 ($PREFIX/share/cartographer_ros_msgs/cmake)
 │ │ CMake Warning (dev) at $PREFIX/share/pcl-1.14/PCLConfig.cmake:121 (find_package):
 │ │   Policy CMP0167 is not set: The FindBoost module is removed.  Run "cmake
 │ │   --help-policy CMP0167" for policy details.  Use the cmake_policy command to
 │ │   set the policy and suppress this warning.
 │ │ Call Stack (most recent call first):
 │ │   $PREFIX/share/pcl-1.14/PCLConfig.cmake:294 (find_boost)
 │ │   $PREFIX/share/pcl-1.14/PCLConfig.cmake:559 (find_external_library)
 │ │   CMakeLists.txt:37 (find_package)
 │ │ This warning is for project developers.  Use -Wno-dev to suppress it.
 │ │ -- Found Boost: $PREFIX/lib/cmake/Boost-1.86.0/BoostConfig.cmake (found suitable version "1.86.0", minimum required is "1.65.0") found components: system iostreams filesystem
 │ │ -- looking for PCL_COMMON
 │ │ -- Found PCL_COMMON: $PREFIX/lib/libpcl_common.dylib
 │ │ -- Found PCL: pcl_common;Boost::system;Boost::iostreams;Boost::filesystem
 │ │ -- Found nav_msgs: 5.3.6 ($PREFIX/share/nav_msgs/cmake)
 │ │ -- Found pcl_conversions: 2.6.2 ($PREFIX/share/pcl_conversions/cmake)
 │ │ -- Found rmw_implementation_cmake: 7.3.2 ($PREFIX/share/rmw_implementation_cmake/cmake)
 │ │ -- Found rmw_fastrtps_cpp: 8.4.2 ($PREFIX/share/rmw_fastrtps_cpp/cmake)
 │ │ -- Found OpenSSL: $PREFIX/lib/libcrypto.dylib (found version "3.5.0")
 │ │ -- Found FastRTPS: $PREFIX/include (Required is at least version "2.13")
 │ │ -- Using RMW implementation 'rmw_fastrtps_cpp' as default
 │ │ -- Found rosbag2_cpp: 0.26.6 ($PREFIX/share/rosbag2_cpp/cmake)
 │ │ -- Setting yaml-cpp_DIR to: '$PREFIX/share/yaml_cpp_vendor/cmake/../../../opt/yaml_cpp_vendor/share/cmake/yaml-cpp'
 │ │ -- Found tf2: 0.36.9 ($PREFIX/share/tf2/cmake)
 │ │ -- Found tf2_eigen: 0.36.9 ($PREFIX/share/tf2_eigen/cmake)
 │ │ -- Found urdf: 2.10.0 ($PREFIX/share/urdf/cmake)
 │ │ -- Found visualization_msgs: 5.3.6 ($PREFIX/share/visualization_msgs/cmake)
 │ │ -- Configuring done (6.3s)
 │ │ CMake Error at $PREFIX/share/cartographer/cmake/CartographerTargets.cmake:61 (set_target_properties):
 │ │   The link interface of target "cartographer" contains:
 │ │     protobuf::libprotobuf
 │ │   but the target was not found.  Possible reasons include:
 │ │     * There is a typo in the target name.
 │ │     * A find_package call is missing for an IMPORTED target.
 │ │     * An ALIAS target is missing.
 │ │ Call Stack (most recent call first):
 │ │   $PREFIX/share/cartographer/cartographer-config.cmake:71 (include)
 │ │   CMakeLists.txt:32 (find_package)
 │ │ -- Generating done (0.3s)
 │ │ CMake Warning:
 │ │   Manually-specified variables were not used by the project:
 │ │     AMENT_PREFIX_PATH
 │ │     CATKIN_SKIP_TESTING
 │ │     CMAKE_INSTALL_LIBDIR
 │ │     CMAKE_INSTALL_SYSTEM_RUNTIME_LIBS_SKIP
 │ │     PYTHON_EXECUTABLE
 │ │     PYTHON_INSTALL_DIR
 │ │     Python_EXECUTABLE
 │ │     SETUPTOOLS_DEB_LAYOUT
 │ │ CMake Generate step failed.  Build files cannot be regenerated correctly.
 │ │ × error Script failed with status 1
 │ │ × error Work directory: '/Users/runner/conda-bld/bld/rattler-build_ros-jazzy-cartographer-ros_1744712279/work'
 │ │ × error To debug the build, run it manually in the work directory (execute the `./conda_build.sh` or `conda_build.bat` script)
~~~

closed 2025-04-17 by Tobias-Fischer