RoboStack/ros-jazzy #44: Build on main branch broken [windows]

Tobias-Fischer, 2025-04-14:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Uploads fail with:
```
Pixi task (upload in default): anaconda -t $ANACONDA_API_TOKEN upload C:\bld\win-64\*.conda --force
Traceback (most recent call last):
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Scripts\anaconda-script.py", line 6, in <module>
    from binstar_client.scripts.cli import main
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\binstar_client\__init__.py", line 23, in <module>
    from .mixins.channels import ChannelsMixin
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\binstar_client\mixins\channels.py", line 9, in <module>
    from binstar_client.utils import jencode
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\binstar_client\utils\__init__.py", line 13, in <module>
    from .config import (
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\binstar_client\utils\config.py", line 109, in <module>
    dirs.site_data_dir,
    ^^^^^^^^^^^^^^^^^^
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\platformdirs\windows.py", line 56, in site_data_dir
    path = os.path.normpath(get_win_folder("CSIDL_COMMON_APPDATA"))
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\platformdirs\windows.py", line 209, in get_win_folder_from_registry
    directory, _ = winreg.QueryValueEx(key, shell_folder_name)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [WinError 2] The system cannot find the file specified
```

### Installed packages

```shell
n/a
```

### Environment info

```shell
n/a
```

Tobias-Fischer, 2025-04-14:
Some possibly related issues:
- https://github.com/Azure/azure-cli/issues/29630
- https://github.com/orgs/python-poetry/discussions/9807
- https://github.com/conda/conda/issues/8691

traversaro, 2025-04-16:
https://github.com/anaconda/anaconda-client/issues/399

traversaro, 2025-04-16:
Great, I reproduced the issue locally.

traversaro, 2025-04-16:
I was able to fix the problem by refreshing the pixi.lock, so probably updating the anaconda-client or another library is sufficient, I guess the problem was triggered by the update to Python 3.12 .

traversaro, 2025-04-16:
I was able to hot-fix the problem by refreshing the pixi.lock (https://github.com/RoboStack/ros-jazzy/commit/84322883a407d7e47981e72611b4ebfb00526814) so probably updating the anaconda-client or another library is sufficient, I guess the problem was triggered by the update to Python 3.12 . We need to do the same fix in the main branch to ensure that future builds are also fixed.

traversaro, 2025-04-18:
Still broken:

~~~
Pixi task (upload in default): anaconda -t $ANACONDA_API_TOKEN upload C:\bld\win-64\*.conda --force
Traceback (most recent call last):
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Scripts\anaconda-script.py", line 6, in <module>
    from anaconda_cli_base.cli import app
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\anaconda_cli_base\__init__.py", line 6, in <module>
    from anaconda_cli_base.console import init_logging, console
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\anaconda_cli_base\console.py", line 5, in <module>
    import click
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\click\__init__.py", line 8, in <module>
    from .core import Argument as Argument
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\click\core.py", line 16, in <module>
    from . import types
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\click\types.py", line 9, in <module>
    from ._compat import _get_argv_encoding
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\click\_compat.py", line 510, in <module>
    from ._winconsole import _get_windows_console_stream
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\site-packages\click\_winconsole.py", line 13, in <module>
    from ctypes import byref
  File "D:\a\ros-jazzy\ros-jazzy\.pixi\envs\default\Lib\ctypes\__init__.py", line 8, in <module>
    from _ctypes import Union, Structure, Array
ImportError: DLL load failed while importing _ctypes: The specified procedure could not be found.
~~~

From https://github.com/RoboStack/ros-jazzy/actions/runs/14513410145 . The problem is that I tried locally by removing all the `pixi.lock` and it started working, but then in the PR I just update the constraint just on two libraries. We should have a smoke check on anaconda upload working fine in the PR instead of discovering it post-PR.

closed 2025-04-18 by traversaro