RoboStack/ros-jazzy #49: Rviz2 cannot startup on macOS M4

Sefibrah, 2025-04-19:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

It just stops... And I don't have an explanation. It stops when it reaches `Mesh: Loading rviz_cone.mesh`. I don't get kicked out of the terminal, it just sits there, and wait. I really don't want to use a VM to work with ROS2, but rather, I want it to work on my macOS M4. So, here's the Terminal log:

```
(ros_env) ➜  ~ rviz2 -l
[INFO] [1745047128.454337000] [rviz2]: Creating resource group General
[INFO] [1745047128.454394000] [rviz2]: Creating resource group OgreInternal
[INFO] [1745047128.454403000] [rviz2]: Creating resource group OgreAutodetect
[INFO] [1745047128.454414000] [rviz2]: SceneManagerFactory for type 'DefaultSceneManager' registered.
[INFO] [1745047128.454441000] [rviz2]: Registering ResourceManager for type Material
[INFO] [1745047128.454446000] [rviz2]: Registering ResourceManager for type Mesh
[INFO] [1745047128.454450000] [rviz2]: Registering ResourceManager for type Skeleton
[INFO] [1745047128.454454000] [rviz2]: MovableObjectFactory for type 'ParticleSystem' registered.
[INFO] [1745047128.454468000] [rviz2]: ArchiveFactory for archive type FileSystem registered.
[INFO] [1745047128.454473000] [rviz2]: ArchiveFactory for archive type Zip registered.
[INFO] [1745047128.454476000] [rviz2]: ArchiveFactory for archive type EmbeddedZip registered.
[INFO] [1745047128.454480000] [rviz2]: DDS codec registering
[INFO] [1745047128.454483000] [rviz2]: ETC codec registering
[INFO] [1745047128.454487000] [rviz2]: ASTC codec registering
[INFO] [1745047128.454490000] [rviz2]: Registering ResourceManager for type HighLevelGpuProgram
[INFO] [1745047128.454494000] [rviz2]: Registering ResourceManager for type Compositor
[INFO] [1745047128.454528000] [rviz2]: MovableObjectFactory for type 'Entity' registered.
[INFO] [1745047128.454533000] [rviz2]: MovableObjectFactory for type 'Light' registered.
[INFO] [1745047128.454537000] [rviz2]: MovableObjectFactory for type 'BillboardSet' registered.
[INFO] [1745047128.454541000] [rviz2]: MovableObjectFactory for type 'ManualObject' registered.
[INFO] [1745047128.454545000] [rviz2]: MovableObjectFactory for type 'BillboardChain' registered.
[INFO] [1745047128.454550000] [rviz2]: MovableObjectFactory for type 'RibbonTrail' registered.
[INFO] [1745047128.454595000] [rviz2]: *-*-* OGRE Initialising
[INFO] [1745047128.454601000] [rviz2]: *-*-* Version 1.12.10 (Rhagorthua)
[INFO] [1745047128.454623000] [rviz2]: OverlayElementFactory for type Panel registered.
[INFO] [1745047128.454628000] [rviz2]: OverlayElementFactory for type BorderPanel registered.
[INFO] [1745047128.454633000] [rviz2]: OverlayElementFactory for type TextArea registered.
[INFO] [1745047128.454637000] [rviz2]: Registering ResourceManager for type Font
[INFO] [1745047128.454644000] [rviz2]: Loading library /Users/ibrahimsefer/miniconda3/envs/ros_env/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.dylib
[INFO] [1745047128.455975000] [rviz2]: Installing plugin: GL RenderSystem
[INFO] [1745047128.455998000] [rviz2]: OpenGL Rendering Subsystem created.
[INFO] [1745047128.461380000] [rviz2]: Plugin successfully installed
[INFO] [1745047128.461411000] [rviz2]: Loading library /Users/ibrahimsefer/miniconda3/envs/ros_env/opt/rviz_ogre_vendor/lib/OGRE/Codec_STBI.dylib
[INFO] [1745047128.462011000] [rviz2]: stb_image - v2.23 - public domain image loader
[INFO] [1745047128.462026000] [rviz2]: Supported formats: jpeg,jpg,png,bmp,psd,tga,gif,pic,ppm,pgm,hdr
[INFO] [1745047128.462038000] [rviz2]: CPU Identifier & Features
[INFO] [1745047128.462042000] [rviz2]: -------------------------
[INFO] [1745047128.462046000] [rviz2]:  *   CPU ID: 
[INFO] [1745047128.462049000] [rviz2]:  *          VFP: no
[INFO] [1745047128.462053000] [rviz2]:  *         NEON: yes
[INFO] [1745047128.462056000] [rviz2]: -------------------------
[INFO] [1745047128.462059000] [rviz2]: ********************************************
***  Starting Mac OS X OpenGL Subsystem  ***
********************************************
[INFO] [1745047128.462064000] [rviz2]: Registering ResourceManager for type Texture
[INFO] [1745047128.462081000] [rviz2]: RenderSystem::_createRenderWindow "OgreWindow(0)", 1x1 windowed  miscParams: FSAA=4 border=none contentScalingFactor=1.000000 currentGLContext=false left=0 macAPI=cocoa macAPICocoaUseNSView=true parentWindowHandle=0 top=0 
[INFO] [1745047128.462088000] [rviz2]: Creating a Cocoa Compatible Render System
[INFO] [1745047128.465847000] [rviz2]: Cocoa: Window created 1 x 1 with backing store size 2 x 2 using content scaling factor 2.0
[INFO] [1745047128.479028000] [rviz2]: GL_VERSION = 2.1.0.0
[INFO] [1745047128.479070000] [rviz2]: GL_VENDOR = Apple
[INFO] [1745047128.479077000] [rviz2]: GL_RENDERER = Apple M4 Max
[INFO] [1745047128.479084000] [rviz2]: GL_EXTENSIONS = GL_ARB_color_buffer_float GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_draw_buffers GL_ARB_draw_elements_base_vertex GL_ARB_draw_instanced GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_fragment_shader GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_imaging GL_ARB_instanced_arrays GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_pixel_buffer_object GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_provoking_vertex GL_ARB_seamless_cube_map GL_ARB_shader_objects GL_ARB_shader_texture_lod GL_ARB_shading_language_100 GL_ARB_shadow GL_ARB_shadow_ambient GL_ARB_sync GL_ARB_texture_border_clamp GL_ARB_texture_compression GL_ARB_texture_compression_rgtc GL_ARB_texture_cube_map GL_ARB_texture_env_add GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_float GL_ARB_texture_mirrored_repeat GL_ARB_texture_non_power_of_two GL_ARB_texture_rectangle GL_ARB_texture_rg GL_ARB_transpose_matrix GL_ARB_vertex_array_bgra GL_ARB_vertex_blend GL_ARB_vertex_buffer_object GL_ARB_vertex_program GL_ARB_vertex_shader GL_ARB_window_pos GL_EXT_abgr GL_EXT_bgra GL_EXT_bindable_uniform GL_EXT_blend_color GL_EXT_blend_equation_separate GL_EXT_blend_func_separate GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_clip_volume_hint GL_EXT_debug_label GL_EXT_debug_marker GL_EXT_draw_buffers2 GL_EXT_draw_range_elements GL_EXT_fog_coord GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_framebuffer_object GL_EXT_framebuffer_sRGB GL_EXT_geometry_shader4 GL_EXT_gpu_program_parameters GL_EXT_gpu_shader4 GL_EXT_multi_draw_arrays GL_EXT_packed_depth_stencil GL_EXT_packed_float GL_EXT_provoking_vertex GL_EXT_rescale_normal GL_EXT_secondary_color GL_EXT_separate_specular_color GL_EXT_shadow_funcs GL_EXT_stencil_two_side GL_EXT_stencil_wrap GL_EXT_texture_array GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_s3tc GL_EXT_texture_env_add GL_EXT_texture_filter_anisotropic GL_EXT_texture_integer GL_EXT_texture_lod_bias GL_EXT_texture_rectangle GL_EXT_texture_shared_exponent GL_EXT_texture_sRGB GL_EXT_texture_sRGB_decode GL_EXT_timer_query GL_EXT_transform_feedback GL_EXT_vertex_array_bgra GL_APPLE_aux_depth_stencil GL_APPLE_client_storage GL_APPLE_element_array GL_APPLE_fence GL_APPLE_float_pixels GL_APPLE_flush_buffer_range GL_APPLE_flush_render GL_APPLE_packed_pixels GL_APPLE_pixel_buffer GL_APPLE_rgb_422 GL_APPLE_row_bytes GL_APPLE_specular_vector GL_APPLE_texture_range GL_APPLE_transform_hint GL_APPLE_vertex_array_object GL_APPLE_vertex_point_size GL_APPLE_vertex_program_evaluators GL_APPLE_ycbcr_422 GL_ATI_separate_stencil GL_ATI_texture_env_combine3 GL_ATI_texture_float GL_IBM_rasterpos_clip GL_NV_blend_square GL_NV_conditional_render GL_NV_depth_clamp GL_NV_fog_distance GL_NV_fragment_program_option GL_NV_fragment_program2 GL_NV_light_max_exponent GL_NV_texgen_reflection GL_NV_texture_barrier GL_NV_vertex_program2_option GL_NV_vertex_program3 GL_SGI_color_matrix GL_SGIS_generate_mipmap GL_SGIS_texture_edge_clamp GL_SGIS_texture_lod 
[INFO] [1745047128.479246000] [rviz2]: ***************************
[INFO] [1745047128.479256000] [rviz2]: *** GL Renderer Started ***
[INFO] [1745047128.479261000] [rviz2]: ***************************
[INFO] [1745047128.496986000] [rviz2]: Registering ResourceManager for type GpuProgram
[INFO] [1745047128.497040000] [rviz2]: GLSL support detected
[INFO] [1745047128.497047000] [rviz2]: GL: Using GL_EXT_framebuffer_object for rendering to textures (best)
[... cut ...]
          package cache : /Users/ibrahimsefer/miniconda3/pkgs
                          /Users/ibrahimsefer/.conda/pkgs
       envs directories : /Users/ibrahimsefer/miniconda3/envs
                          /Users/ibrahimsefer/.conda/envs
               platform : osx-arm64
             user-agent : conda/25.3.1 requests/2.32.3 CPython/3.12.2 Darwin/24.3.0 OSX/15.3.1 solver/libmamba aau/0.5.0 c/. s/. e/.
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

wep21, 2025-04-21:
What happens if you change dds like
```
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp rviz2 -l
```
?

KeqiZeng, 2025-04-22:
> What happens if you change dds like
> 
> ```
> RMW_IMPLEMENTATION=rmw_cyclonedds_cpp rviz2 -l
> ```
> 
> ?

HI, 

I encountered the same issue and when I ran `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp rviz2 -l`, `rviz2` can startup successfully!

My system: macOS Sequoia 15.3.2 arm64

Sefibrah, 2025-04-25:
@wep21 I tried your suggestion and it worked!

But why did I have to use `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`? 

And another question, will this RMW thing limit some capabilities in my rviz2 now?

Tobias-Fischer, 2025-04-26:
Let’s leave this open till it’s properly fixed

wep21, 2025-04-26:
@Sefibrah 

> But why did I have to use RMW_IMPLEMENTATION=rmw_cyclonedds_cpp?

Currently, it doesn't work properly with fastrtps which is default dds as of ros2 jazzy on macos.
I confirmed it works with fastrtps on linux.
I'm not sure this is due to conda enviroment which pixi created.

> And another question, will this RMW thing limit some capabilities in my rviz2 now?

I guess it will not. cyclonedds was default dds as of ros2 galactic, it seems to work stable.

h3ct0r, 2025-07-17:
> Does `ros2 run rviz2 rviz2` works? If yes, I suspect this is similar to [#57](https://github.com/RoboStack/ros-jazzy/issues/57) .

I'm having the same issue as OP, and running `ros2 run rviz2 rviz2` does not fix the issue. I was able to run it successfully with `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp rviz2 -l` instead. Even commands such as `ros2 topic list` just hang without specifying `rmw_cyclonedds_cpp`.

I'm using ros jazzy on Mac M3 (fresh mamba install).

Tobias-Fischer, 2025-07-27:
See #50

closed 2025-07-27 by Tobias-Fischer

traversaro, 2025-07-28:
> [@traversaro](https://github.com/traversaro) did we move to rmw_cyclonedds_cpp on macOS and can close here?

I am afraid we did not include the patch https://github.com/RoboStack/ros-humble/blob/main/patch/ros-humble-rmw-implementation-cmake.osx.patch in the last rebuild. We could probably add it without a full rebuild by building only the affected package, but it would be convenient to have some kind of tests of the fastrtps/macos problem (ideally without involving rviz2, something like https://github.com/RoboStack/ros-jazzy/blob/main/tests/ros-jazzy-robot-state-publisher.yaml and https://github.com/RoboStack/ros-jazzy/blob/main/tests/ros-jazzy-robot-state-publisher/robot_state_publisher_smoke_test_launch.py) to avoid regressions.