RoboStack/ros-jazzy #50: Failed to subscribe ROS 2 topic with pixi and ROS 2 Jazzy

atinfinity, 2025-04-19:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

I failed to subscribe ROS 2 topic with pixi and ROS 2 Jazzy.

### Reproducible example

I installed `ros-jazzy-desktop` by pixi.

```bash
pixi init my_ros2_project -c robostack-jazzy -c conda-forge
cd my_ros2_project
pixi add ros-jazzy-desktop
```

And, I published topic. 

```bash
pixi run ros2 topic pub --rate 1 /turtle1/cmd_vel geometry_msgs/msg/Twist "{linear: {x: 2.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.8}}"
```

But, I could NOT subscribe topic.

```bash
pixi run ros2 topic echo /turtle1/cmd_vel
```

My environment is as follows.

- macOS Sonoma 14.4
- pixi 0.45.0

### Issue description

Failed to subscribe ROS 2 topic with pixi and ROS 2 Jazzy.

```bash
pixi run ros2 topic pub --rate 1 /turtle1/cmd_vel geometry_msgs/msg/Twist "{linear: {x: 2.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.8}}"
```

```bash
pixi run ros2 topic echo /turtle1/cmd_vel
```

### Expected behavior

I expected that I can subscribe `/turtle1/cmd_vel`.

### Installed packages

```shell
The result of `pixi list` longer than 65536 characters.
So, I upload to <https://gist.github.com/atinfinity/6d60e56f49e8ba6d742c07e3286dc669>.
```

### Environment info

```shell
System
------------
       Pixi version: 0.45.0
           Platform: osx-arm64
   Virtual packages: __unix=0=0
                   : __osx=14.4=0
                   : __archspec=1=m1
          Cache dir: /Users/dandelion/Library/Caches/rattler/cache
       Auth storage: /Users/dandelion/.rattler/credentials.json
   Config locations: No config files found

Global
------------
            Bin dir: /Users/dandelion/.pixi/bin
    Environment dir: /Users/dandelion/.pixi/envs
       Manifest dir: /Users/dandelion/.pixi/manifests/pixi-global.toml

Workspace
------------
               Name: my_ros2_project
            Version: 0.1.0
      Manifest file: /Users/dandelion/my_ros2_project/pixi.toml
       Last updated: 19-04-2025 10:23:12

Environments
------------
        Environment: default
           Features: default
           Channels: robostack-jazzy, conda-forge
   Dependency count: 1
       Dependencies: ros-jazzy-desktop
   Target platforms: osx-arm64
```

wep21, 2025-04-20:
What happens if you add `--no-daemon` into the ros2 topic command?

atinfinity, 2025-04-20:
I add `--no-daemon` to `ros2 topic echo`. And, I didn't add `ros2 topic pub`. Because, there is no `--no-daemon` option.

```bash
pixi run ros2 topic pub --rate 1 /turtle1/cmd_vel geometry_msgs/msg/Twist "{linear: {x: 2.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.8}}"
pixi run ros2 topic echo /turtle1/cmd_vel --no-daemon
```

As a result, I failed to subscribe ROS 2 topic .

wep21, 2025-04-20:
Is it possible to check the behavior again after `ros2 daemon stop` and `ros2 daemon start`?

atinfinity, 2025-04-20:
I tried it.

```bash
$ pixi run ros2 daemon stop
The daemon is not running
```

```bash
$ pixi run ros2 daemon start
The daemon has been started
```

```bash
$ pixi run ros2 topic pub --rate 1 /turtle1/cmd_vel geometry_msgs/msg/Twist "{linear: {x: 2.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.8}}"
publisher: beginning loop
publishing #1: geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=2.0, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=1.8))
```

```bash
pixi run ros2 topic echo /turtle1/cmd_vel
WARNING: topic [/turtle1/cmd_vel] does not appear to be published yet
```

atinfinity, 2025-04-20:
I used Cyclone DDS. In this case, there is no problem.

```bash
pixi init my_ros2_project -c robostack-jazzy -c conda-forge
cd my_ros2_project
pixi add ros-jazzy-desktop ros-jazzy-rmw-cyclonedds-cpp
touch activate.sh
```

I add the following description to `activate.sh`.

```bash
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
```

I add the following description to `pixi.toml`.

```toml
[activation]
scripts = ["activate.sh"]
```

```bash
$ pixi run ros2 topic pub --rate 1 /turtle1/cmd_vel geometry_msgs/msg/Twist "{linear: {x: 2.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 1.8}}"
publisher: beginning loop
publishing #1: geometry_msgs.msg.Twist(linear=geometry_msgs.msg.Vector3(x=2.0, y=0.0, z=0.0), angular=geometry_msgs.msg.Vector3(x=0.0, y=0.0, z=1.8))
```

```bash
$ pixi run ros2 topic echo /turtle1/cmd_vel
linear:
  x: 2.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 1.8
```

wep21, 2025-04-20:
I can reproduce the issue with fastdds on macos. There is no issue with cyclonedds. On the other hand, neither dds on linux causes problems.

wep21, 2025-06-18:
ROS 2 communication by fastdds on macos works fine with ros-kilted, though...

```
❯ pixi run ros2 run demo_nodes_cpp talker
Problem to set affinity of thread with id [0x16b37b000,dds.log] to value 0. Error 'Protocol family not supported'
2025-06-18 19:18:49.332 [SYSTEM Error] Problem to set affinity of thread with id [0x16b2ef000,dds.shm.wdog] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-06-18 19:18:49.333 [SYSTEM Error] Problem to set affinity of thread with id [0x16b407000,dds.ev.0] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-06-18 19:18:49.333 [SYSTEM Error] Problem to set affinity of thread with id [0x16b51f000,dds.udp.7400] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-06-18 19:18:49.333 [SYSTEM Error] Problem to set affinity of thread with id [0x16b493000,dds.udp.7410] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-06-18 19:18:49.334 [SYSTEM Error] Problem to set affinity of thread with id [0x16b5ab000,dds.shm.7411] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-06-18 19:18:49.334 [SYSTEM Error] Problem to set affinity of thread with id [0x16b637000,dds.udp.7411] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-06-18 19:18:49.334 [SYSTEM Error] Problem to set affinity of thread with id [0x16b6c3000,dds.asyn.0.0] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-06-18 19:18:49.334 [SYSTEM Error] Problem to set affinity of thread with id [0x16b74f000,dds.tls.request] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-06-18 19:18:49.334 [SYSTEM Error] Problem to set affinity of thread with id [0x16b7db000,dds.asyn.0.1] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-06-18 19:18:49.334 [SYSTEM Error] Problem to set affinity of thread with id [0x16b867000,dds.tls.replies] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-06-18 19:18:49.340 [SYSTEM Error] Problem to set affinity of thread with id [0x16b97f000,dds.udp.7000] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-06-18 19:18:49.340 [SYSTEM Error] Problem to set affinity of thread with id [0x16b8f3000,dds.shm.7000] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
[INFO] [1750241930.390874000] [talker]: Publishing: 'Hello World: 1'
[INFO] [1750241931.386822000] [talker]: Publishing: 'Hello World: 2'
```

Tobias-Fischer, 2025-07-27:
We now use cyclonedds by default; so this should be mostly a non-issue hopefully.

traversaro, 2025-09-08:
> Running `echo $RMW_IMPLEMENTATION` doesn't return anything, from my search I understand this means that the environment is using the default dds (fastDDS).

Sorry for the misunderstanding (I think someone on Discord predicted that would be confusing), but if the env variable is not set, it just means that the default is used, and we changed the default in macos to `cyclonedds` in https://github.com/RoboStack/ros-jazzy/pull/95, see https://github.com/RoboStack/ros-jazzy/blob/main/patch/ros-jazzy-rwm-implementation-cmake.osx.patch . So, to check you actual default rmw you need to check that file.

TedVanderfeen, 2025-09-09:
Just trying to wrap my head around the build system, but should there be osx specific build targets in the [vinca.yaml](https://github.com/RoboStack/ros-jazzy/blob/main/vinca.yaml) in order for it to apply these patches?

Tobias-Fischer, 2025-09-09:
Patches are applied to all platforms on default, unless there is a platform identifier before the `.patch` suffix (as in https://github.com/RoboStack/ros-jazzy/blob/main/patch/ros-jazzy-rwm-implementation-cmake.osx.patch for example)

atinfinity, 2025-11-04:
Thank you for your sharing of information.

> There is stil a signal error when closing rviz2

I could reproduce this error.

```bash
pixi run rviz2
Problem to set affinity of thread with id [0x16fa8f000,dds.log] to value 0. Error 'Protocol family not supported'
2025-11-05 00:49:27.727 [SYSTEM Error] Problem to set affinity of thread with id [0x16fa03000,dds.shm.wdog] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-11-05 00:49:27.728 [SYSTEM Error] Problem to set affinity of thread with id [0x16fb1b000,dds.ev.0] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-11-05 00:49:27.728 [SYSTEM Error] Problem to set affinity of thread with id [0x16fba7000,dds.udp.7410] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-11-05 00:49:27.728 [SYSTEM Error] Problem to set affinity of thread with id [0x16fc33000,dds.udp.7400] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-11-05 00:49:27.728 [SYSTEM Error] Problem to set affinity of thread with id [0x16fd4b000,dds.udp.7411] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-11-05 00:49:27.728 [SYSTEM Error] Problem to set affinity of thread with id [0x16fcbf000,dds.shm.7411] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-11-05 00:49:27.728 [SYSTEM Error] Problem to set affinity of thread with id [0x16fdd7000,dds.asyn.0.0] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-11-05 00:49:27.732 [SYSTEM Error] Problem to set affinity of thread with id [0x16fe63000,dds.shm.7000] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
2025-11-05 00:49:27.732 [SYSTEM Error] Problem to set affinity of thread with id [0x16feef000,dds.udp.7000] to value 0. Error 'Protocol family not supported' -> Function configure_current_thread_affinity
```

It seems that https://github.com/eProsima/Fast-DDS/issues/5524 is related report.