RoboStack/ros-jazzy #56: RVIZ2 on Mac OS M2 Sequoia

ipolytex, 2025-05-18:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

After installed ROS2 on MAC OS M2 (SEQUIOA) the startup screen freeze at initializing.

rviz2 -l
[INFO] [1747497126.394856000] [rviz2]: Creating resource group General
[INFO] [1747497126.394915000] [rviz2]: Creating resource group OgreInternal
[INFO] [1747497126.394926000] [rviz2]: Creating resource group OgreAutodetect
[INFO] [1747497126.394938000] [rviz2]: SceneManagerFactory for type 'DefaultSceneManager' registered.
[INFO] [1747497126.394967000] [rviz2]: Registering ResourceManager for type Material
[INFO] [1747497126.394976000] [rviz2]: Registering ResourceManager for type Mesh
[INFO] [1747497126.394982000] [rviz2]: Registering ResourceManager for type Skeleton
[INFO] [1747497126.394989000] [rviz2]: MovableObjectFactory for type 'ParticleSystem' registered.
[INFO] [1747497126.395002000] [rviz2]: ArchiveFactory for archive type FileSystem registered.
[INFO] [1747497126.395009000] [rviz2]: ArchiveFactory for archive type Zip registered.
[INFO] [1747497126.395015000] [rviz2]: ArchiveFactory for archive type EmbeddedZip registered.
[INFO] [1747497126.395020000] [rviz2]: DDS codec registering
[INFO] [1747497126.395025000] [rviz2]: ETC codec registering
[INFO] [1747497126.395031000] [rviz2]: ASTC codec registering
[INFO] [1747497126.395036000] [rviz2]: Registering ResourceManager for type HighLevelGpuProgram
[INFO] [1747497126.395043000] [rviz2]: Registering ResourceManager for type Compositor
[INFO] [1747497126.395077000] [rviz2]: MovableObjectFactory for type 'Entity' registered.
[INFO] [1747497126.395086000] [rviz2]: MovableObjectFactory for type 'Light' registered.
[INFO] [1747497126.395091000] [rviz2]: MovableObjectFactory for type 'BillboardSet' registered.
[INFO] [1747497126.395097000] [rviz2]: MovableObjectFactory for type 'ManualObject' registered.
[INFO] [1747497126.395102000] [rviz2]: MovableObjectFactory for type 'BillboardChain' registered.
[INFO] [1747497126.395108000] [rviz2]: MovableObjectFactory for type 'RibbonTrail' registered.
[INFO] [1747497126.395164000] [rviz2]: --* OGRE Initialising
[INFO] [1747497126.395173000] [rviz2]: --* Version 1.12.10 (Rhagorthua)
[INFO] [1747497126.395192000] [rviz2]: OverlayElementFactory for type Panel registered.
[INFO] [1747497126.395199000] [rviz2]: OverlayElementFactory for type BorderPanel registered.
[INFO] [1747497126.395204000] [rviz2]: OverlayElementFactory for type TextArea registered.
[INFO] [1747497126.395210000] [rviz2]: Registering ResourceManager for type Font
[INFO] [1747497126.395217000] [rviz2]: Loading library /Users/adrianchan/miniconda3/envs/ros2/opt/rviz_ogre_vendor/lib/OGRE/RenderSystem_GL.dylib
[INFO] [1747497126.397735000] [rviz2]: Installing plugin: GL RenderSystem
[INFO] [1747497126.397757000] [rviz2]: OpenGL Rendering Subsystem created.
[INFO] [1747497126.408472000] [rviz2]: Plugin successfully installed
[INFO] [1747497126.408506000] [rviz2]: Loading library /Users/adrianchan/miniconda3/envs/ros2/opt/rviz_ogre_vendor/lib/OGRE/Codec_STBI.dylib
[INFO] [1747497126.410192000] [rviz2]: stb_image - v2.23 - public domain image loader
[INFO] [1747497126.410211000] [rviz2]: Supported formats: jpeg,jpg,png,bmp,psd,tga,gif,pic,ppm,pgm,hdr
[INFO] [1747497126.410229000] [rviz2]: CPU Identifier & Features
[INFO] [1747497126.410237000] [rviz2]: -------------------------
[INFO] [1747497126.410244000] [rviz2]: * CPU ID:
[INFO] [1747497126.410250000] [rviz2]: * VFP: no
[INFO] [1747497126.410256000] [rviz2]: * NEON: yes
[INFO] [1747497126.410262000] [rviz2]: -------------------------
[INFO] [1747497126.410269000] [rviz2]: ********************************************
*** Starting Mac OS X OpenGL Subsystem ***

[INFO] [1747497126.410277000] [rviz2]: Registering ResourceManager for type Texture
[INFO] [1747497126.410297000] [rviz2]: RenderSystem::_createRenderWindow "OgreWindow(0)", 1x1 windowed miscParams: FSAA=4 border=none contentScalingFactor=1.000000 currentGLContext=false left=0 macAPI=cocoa macAPICocoaUseNSView=true parentWindowHandle=0 top=0
[INFO] [1747497126.410307000] [rviz2]: Creating a Cocoa Compatible Render System
[INFO] [1747497126.415660000] [rviz2]: Cocoa: Window created 1 x 1 with backing store size 2 x 2 using content scaling factor 2.0
[INFO] [1747497126.418714000] [rviz2]: GL_VERSION = 2.1.0.0
[INFO] [1747497126.418735000] [rviz2]: GL_VENDOR = Apple
[INFO] [1747497126.418744000] [rviz2]: GL_RENDERER = Apple M2 Max
[INFO] [1747497126.418757000] [rviz2]: GL_EXTENSIONS = GL_ARB_color_buffer_float GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_draw_buffers GL_ARB_draw_elements_base_vertex GL_ARB_draw_instanced GL_ARB_fragment_program GL_ARB_fragment_program_shadow GL_ARB_fragment_shader GL_ARB_framebuffer_object GL_ARB_framebuffer_sRGB GL_ARB_half_float_pixel GL_ARB_half_float_vertex GL_ARB_imaging GL_ARB_instanced_arrays GL_ARB_multisample GL_ARB_multitexture GL_ARB_occlusion_query GL_ARB_pixel_buffer_object GL_ARB_point_parameters GL_ARB_point_sprite GL_ARB_provoking_vertex GL_ARB_seamless_cube_map GL_ARB_shader_objects GL_ARB_shader_texture_lod GL_ARB_shading_language_100 GL_ARB_shadow GL_ARB_shadow_ambient GL_ARB_sync GL_ARB_texture_border_clamp GL_ARB_texture_compression GL_ARB_texture_compression_rgtc GL_ARB_texture_cube_map GL_ARB_texture_env_add GL_ARB_texture_env_combine GL_ARB_texture_env_crossbar GL_ARB_texture_env_dot3 GL_ARB_texture_float GL_ARB_texture_mirrored_repeat GL_ARB_texture_non_power_of_two GL_ARB_texture_rectangle GL_ARB_texture_rg GL_ARB_transpose_matrix GL_ARB_vertex_array_bgra GL_ARB_vertex_blend GL_ARB_vertex_buffer_object GL_ARB_vertex_program GL_ARB_vertex_shader GL_ARB_window_pos GL_EXT_abgr GL_EXT_bgra GL_EXT_bindable_uniform GL_EXT_blend_color GL_EXT_blend_equation_separate GL_EXT_blend_func_separate GL_EXT_blend_minmax GL_EXT_blend_subtract GL_EXT_clip_volume_hint GL_EXT_debug_label GL_EXT_debug_marker GL_EXT_draw_buffers2 GL_EXT_draw_range_elements GL_EXT_fog_coord GL_EXT_framebuffer_blit GL_EXT_framebuffer_multisample GL_EXT_framebuffer_multisample_blit_scaled GL_EXT_framebuffer_object GL_EXT_framebuffer_sRGB GL_EXT_geometry_shader4 GL_EXT_gpu_program_parameters GL_EXT_gpu_shader4 GL_EXT_multi_draw_arrays GL_EXT_packed_depth_stencil GL_EXT_packed_float GL_EXT_provoking_vertex GL_EXT_rescale_normal GL_EXT_secondary_color GL_EXT_separate_specular_color GL_EXT_shadow_funcs GL_EXT_stencil_two_side GL_EXT_stencil_wrap GL_EXT_texture_array GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_s3tc GL_EXT_texture_env_add GL_EXT_texture_filter_anisotropic GL_EXT_texture_integer GL_EXT_texture_lod_bias GL_EXT_texture_rectangle GL_EXT_texture_shared_exponent GL_EXT_texture_sRGB GL_EXT_texture_sRGB_decode GL_EXT_timer_query GL_EXT_transform_feedback GL_EXT_vertex_array_bgra GL_APPLE_aux_depth_stencil GL_APPLE_client_storage GL_APPLE_element_array GL_APPLE_fence GL_APPLE_float_pixels GL_APPLE_flush_buffer_range GL_APPLE_flush_render GL_APPLE_packed_pixels GL_APPLE_pixel_buffer GL_APPLE_rgb_422 GL_APPLE_row_bytes GL_APPLE_specular_vector GL_APPLE_texture_range GL_APPLE_transform_hint GL_APPLE_vertex_array_object GL_APPLE_vertex_point_size GL_APPLE_vertex_program_evaluators GL_APPLE_ycbcr_422 GL_ATI_separate_stencil GL_ATI_texture_env_combine3 GL_ATI_texture_float GL_IBM_rasterpos_clip GL_NV_blend_square GL_NV_conditional_render GL_NV_depth_clamp GL_NV_fog_distance GL_NV_fragment_program_option GL_NV_fragment_program2 GL_NV_light_max_exponent GL_NV_texgen_reflection GL_NV_texture_barrier GL_NV_vertex_program2_option GL_NV_vertex_program3 GL_SGI_color_matrix GL_SGIS_generate_mipmap GL_SGIS_texture_edge_clamp GL_SGIS_texture_lod
[INFO] [1747497126.418882000] [rviz2]: ***************************
[INFO] [1747497126.418892000] [rviz2]: *** GL Renderer Started ***
[INFO] [1747497126.418898000] [rviz2]: ***************************
[INFO] [1747497126.445024000] [rviz2]: Registering ResourceManager for type GpuProgram
[INFO] [1747497126.445068000] [rviz2]: GLSL support detected
[INFO] [1747497126.445076000] [rviz2]: GL: Using GL_EXT_framebuffer_object for rendering to textures (best)
[INFO] [1747497126.445308000] [rviz2]: FBO PF_UNKNOWN depth/stencil support: D0S1 D0S4 D0S8 D0S16 D16S0 D16S1 D16S4 D16S8 D16S16 D24S0 D24S1 D24S4 D24S8 D24S16 D32S0 D32S1 D32S4 D32S8 D32S16 Packed-D24S8
[... cut ...]
          package cache : /Users/adrianchan/miniconda3/pkgs
                          /Users/adrianchan/.conda/pkgs
       envs directories : /Users/adrianchan/miniconda3/envs
                          /Users/adrianchan/.conda/envs
               platform : osx-arm64
             user-agent : conda/25.3.1 requests/2.32.3 CPython/3.13.1 Darwin/24.5.0 OSX/15.5 solver/libmamba conda-libmamba-solver/25.4.0 libmambapy/2.1.1 aau/0.7.0 c/. s/. e/.
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2025-05-30:
Can you please try this? https://github.com/conda-forge/numpy-feedstock/issues/347#issuecomment-2779297248

x-ref https://github.com/RoboStack/ros-noetic/issues/532

Tobias-Fischer, 2025-06-13:
See https://github.com/RoboStack/ros-jazzy/issues/57#issuecomment-2961098585 for a fix/workaround