RoboStack/ros-jazzy #58: Pixi run build on linux errors

gftabor, 2025-06-05:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

https://gist.github.com/gftabor/9fc1f2f85846921efca64b15d1b5d383

```
│ × error Script failed with status 127
 │ │ × error Work directory: '/home/tabor/Documents/ros-jazzy/output/bld/rattler-build_ros-jazzy-livox-ros-driver2_1749139542/work'
 │ │ × error To debug the build, run it manually in the work directory (execute the `./conda_build.sh` or `conda_build.bat` script)
 │ │
 │ ╰─────────────────── (took 0 seconds)
 │
 ╰─────────────────── (took 2 seconds)
Error:   × Script failed

```

The PR here https://github.com/RoboStack/ros-jazzy/pull/53 adds a symlink that is invalid on my fresh linux install. 

### Installed packages

```shell
NA
```

### Environment info

```shell
NA
```

gftabor, 2025-06-05:
Fixed missing detail from code snip. (the important line wasnt in my snip but was in gist)

gftabor, 2025-06-05:
> The target of the symlink should be created by vinca if I am not wrong, is that not happening for you?

[image]

It didnt seem to create the file in the root folder.

traversaro, 2025-06-05:
Thanks, I reproduced the issue. Can you try to run `pixi run generate-recipes` manually before? It seems like the `generate-recipes` task is not execute before the `build` task, even if it is listed as a dependency.

traversaro, 2025-06-05:
I found the issue, `build_additional_recipes` should have a dependency on `generate-recipes`.

traversaro, 2025-06-05:
> I found the issue, `build_additional_recipes` should have a dependency on `generate-recipes`.

Fix provided in https://github.com/RoboStack/ros-jazzy/pull/59 .

gftabor, 2025-06-05:
@wolfv I can't reopen but I have new error

wolfv, 2025-06-06:
@gftabor that sounds like vinca hasn't generated any recipes yet. I think even an empty recipes folder could work.

gftabor, 2025-06-06:
> [@gftabor](https://github.com/gftabor) that sounds like vinca hasn't generated any recipes yet. I think even an empty recipes folder could work.

https://gist.github.com/gftabor/cd9d98d59eb59256a987f2ade63bc85f

tldr

```
git clone https://github.com/RoboStack/ros-jazzy.git
cd .\ros-jazzy\
pixi run build
error: invalid value './recipes' for '--recipe-dir <RECIPE_DIR>': Path './recipes' needs to exist on disk and be a directory
PS C:\Users\tabor\ros-jazzy> mkdir recipes
```
ls shows recipes directory

```
pixi run build
error: invalid value './recipes' for '--recipe-dir <RECIPE_DIR>': Path './recipes' needs to exist on disk and be a directory
```

ls doesn't show recipes directory, something is deleting it mid `pixi run build` command

closed 2025-06-06 by Tobias-Fischer