RoboStack/ros-jazzy #69: Failed to create new install of ROS2 jazzy

A-Hudson-0117, 2025-06-13:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Following instructions on [Getting Started](https://robostack.github.io/GettingStarted.html) on Ubuntu 22.04. When installing ros desktop it fails. Specifically running `mamba install ros-jazzy-desktop`, I get the error: 
``` bash
conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
warning  libmamba 'repo.anaconda.com', a commercial channel hosted by Anaconda.com, is used.
    
warning  libmamba Please make sure you understand Anaconda Terms of Services.
    
warning  libmamba See: https://legal.anaconda.com/policies/en/

Pinned packages:

  - python=3.11

error    libmamba Could not solve for environment specs
    The following packages are incompatible
    ├─ pin on python 3.11.* =* * is installable and it requires
    │  └─ python =3.11 *, which can be installed;
    └─ ros-jazzy-desktop =* * is not installable because there are no viable options
       ├─ ros-jazzy-desktop 0.11.0 would require
       │  ├─ ros-jazzy-rviz-default-plugins =* * with the potential options
       │  │  ├─ ros-jazzy-rviz-default-plugins 14.1.6 would require
       │  │  │  └─ ros-jazzy-rclcpp =* * with the potential options
       │  │  │     ├─ ros-jazzy-rclcpp 28.1.6 would require
       │  │  │     │  └─ ros-jazzy-rosidl-typesupport-cpp =* * with the potential options
       │  │  │     │     ├─ ros-jazzy-rosidl-typesupport-cpp 3.2.2 would require
       │  │  │     │     │  ├─ ros-jazzy-ament-cmake-core =* *, which requires
       │  │  │     │     │  │  └─ ros2-distro-mutex =0.9 jazzy_*, which can be installed;
       │  │  │     │     │  └─ ros2-distro-mutex =0.6 jazzy_*, which conflicts with any installable versions previously reported;
       │  │  │     │     ├─ ros-jazzy-rosidl-typesupport-cpp 3.2.2 would require
       │  │  │     │     │  ├─ ros-jazzy-ament-cmake-core =* *, which can be installed (as previously explained);
       │  │  │     │     │  └─ ros2-distro-mutex =0.7 jazzy_*, which conflicts with any installable versions previously reported;
       │  │  │     │     ├─ ros-jazzy-rosidl-typesupport-cpp 3.2.2 would require
       │  │  │     │     │  └─ python =3.12 *_cpython, which conflicts with any installable versions previously reported;
       │  │  │     │     ├─ ros-jazzy-rosidl-typesupport-cpp 3.2.2 would require
       │  │  │     │     │  ├─ ros-jazzy-ament-cmake-core =* *, which can be installed (as previously explained);
       │  │  │     │     │  └─ ros2-distro-mutex =0.8 jazzy_*, which conflicts with any installable versions previously reported;
       │  │  │     │     └─ ros-jazzy-rosidl-typesupport-cpp 3.2.2 would require
       │  │  │     │        └─ ros2-distro-mutex =0.9 jazzy_*, which can be installed;
       │  │  │     ├─ ros-jazzy-rclcpp 28.1.2 would require
       │  │  │     │  └─ python =3.12 *_cpython, which conflicts with any installable versions previously reported;
       │  │  │     ├─ ros-jazzy-rclcpp 28.1.6 would require
       │  │  │     │  └─ ros2-distro-mutex =0.7 jazzy_*, which conflicts with any installable versions previously reported;
       │  │  │     ├─ ros-jazzy-rclcpp 28.1.8 would require
       │  │  │     │  └─ ros2-distro-mutex =0.8 jazzy_*, which conflicts with any installable versions previously reported;
       │  │  │     └─ ros-jazzy-rclcpp 28.1.9 would require
       │  │  │        └─ ros2-distro-mutex =0.9 jazzy_*, which can be installed;
       │  │  ├─ ros-jazzy-rviz-default-plugins 14.1.11 would require
       │  │  │  └─ ros2-distro-mutex =0.9 jazzy_*, which can be installed;
       │  │  ├─ ros-jazzy-rviz-default-plugins 14.1.7 would require
       │  │  │  └─ ros2-distro-mutex =0.7 jazzy_*, which conflicts with any installable versions previously reported;
       │  │  └─ ros-jazzy-rviz-default-plugins 14.1.8 would require
       │  │     └─ ros2-distro-mutex =0.8 jazzy_*, which conflicts with any installable versions previously reported;
       │  └─ ros2-distro-mutex =0.6 jazzy_*, which conflicts with any installable versions previously reported;
       ├─ ros-jazzy-desktop 0.11.0 would require
       │  ├─ ros-jazzy-ros-base =* * but there are no viable options
       │  │  ├─ ros-jazzy-ros-base 0.11.0 would require
       │  │  │  ├─ ros-jazzy-rosbag2 =* * but there are no viable options
       │  │  │  │  ├─ ros-jazzy-rosbag2 0.26.7 would require
       │  │  │  │  │  ├─ ros-jazzy-rosbag2-storage-default-plugins =* * but there are no viable options
       │  │  │  │  │  │  ├─ ros-jazzy-rosbag2-storage-default-plugins 0.26.7 would require
       │  │  │  │  │  │  │  ├─ ros-jazzy-rosbag2-storage-mcap =* * but there are no viable options
       │  │  │  │  │  │  │  │  ├─ ros-jazzy-rosbag2-storage-mcap 0.26.7 would require
       │  │  │  │  │  │  │  │  │  └─ ros-jazzy-mcap-vendor =* * but there are no viable options
[... 208 lines cut ...]
     libarchive version : libarchive 3.7.7 zlib/1.2.13 liblzma/5.4.6 bz2lib/1.0.8 liblz4/1.9.4 libzstd/1.5.2
       envs directories : <path_to_mamba>/envs
          package cache : <path_to_mamba>/pkgs
                          $HOME/.mamba/pkgs
            environment : ros_test (active)
           env location : <path_to_mamba>/envs/ros_test
      user config files : $HOME/.mambarc
 populated config files : <path_to_mamba>/envs/ros_test/.condarc
                          $HOME/.condarc
       virtual packages : __unix=0=0
                          __linux=6.8.0=0
                          __glibc=2.35=0
                          __archspec=1=x86_64_v3
                          __cuda=12.8=0
               channels : https://conda.anaconda.org/robostack-jazzy/linux-64
                          https://conda.anaconda.org/robostack-jazzy/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
       base environment : <path_to_mamba>/mamba
               platform : linux-64
```

traversaro, 2025-06-13:
Thanks for opening the issue . The problem with the documentation is that it forces you to create an environmental with Python 3.11, but the latest packages are built against Python 3.12 .

A-Hudson-0117, 2025-06-13:
When is the Python 3.11 build being made?

traversaro, 2025-06-14:
Documentation fix in https://github.com/RoboStack/robostack.github.io/pull/91 .

closed 2025-06-14 by Tobias-Fischer