RoboStack/ros-jazzy #70: lib folders not added to path in windows

gftabor, 2025-06-13:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

[image]

Looks like because windows doesn't have LD_LIBRARY_PATH being appended to when you source a workspace nothing is ever indicating lib folders to be searched. In Windows it looks in the normal PATH for dlls that aren't in the same directory as the executable. 
 
```
C:\Users\tabor\robostack>pixi shell -e jazzy3

(robostack:jazzy3) C:\Users\tabor\robostack>ros2 run apriltag_detector apriltag_detector_node
[ros2run]: Process exited with failure 3221225781

(robostack:jazzy3) C:\Users\tabor\robostack>set PATH=%PATH%;C:\Users\tabor\robostack\.pixi\envs\jazzy3\Library\lib

(robostack:jazzy3) C:\Users\tabor\robostack>ros2 run apriltag_detector apriltag_detector_node
[INFO] [1749855409.373433700] [detector]: tag detector type: umich
```

This problem exists for the current build and locally building with colcon. 

While I am showing the errors for apriltag-detector it likely extends to all sorts of other windows builds of packages.

### Installed packages

```shell
https://gist.github.com/gftabor/241b2486ca415bef47c8d2acb5108d62
```

### Environment info

```shell
(robostack:jazzy3) C:\Users\tabor\robostack>pixi info
System
------------
       Pixi version: 0.48.0
           Platform: win-64
   Virtual packages: __win=10.0.26100=0
                   : __cuda=12.9=0
                   : __archspec=1=skylake
          Cache dir: C:\Users\tabor\AppData\Local\rattler\cache
       Auth storage: C:\Users\tabor\.rattler\credentials.json
   Config locations: No config files found

Global
------------
            Bin dir: C:\Users\tabor\.pixi\bin
    Environment dir: C:\Users\tabor\.pixi\envs
       Manifest dir: C:\Users\tabor\.pixi\manifests\pixi-global.toml

Workspace
------------
               Name: robostack
            Version: 0.1.0
      Manifest file: C:\Users\tabor\robostack\pixi.toml
       Last updated: 13-06-2025 18:51:38

Environments
------------
        Environment: default
           Features: default
           Channels: https://fast.prefix.dev/conda-forge
   Dependency count: 7
       Dependencies: python, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\default

        Environment: noetic
           Features: noetic, default
           Channels: https://prefix.dev/robostack-noetic, https://fast.prefix.dev/conda-forge
   Dependency count: 10
       Dependencies: ros-noetic-desktop, catkin_tools, rosdep, python, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\noetic

        Environment: humble
           Features: humble, default
           Channels: https://prefix.dev/robostack-humble, https://fast.prefix.dev/conda-forge
   Dependency count: 11
       Dependencies: ros-humble-desktop, colcon-common-extensions, rosdep, ros-humble-image-view, python, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\humble

        Environment: jazzy
           Features: jazzy, default
           Channels: https://prefix.dev/robostack-jazzy, https://fast.prefix.dev/conda-forge
   Dependency count: 9
       Dependencies: ros-jazzy-desktop-full, python, ros-jazzy-apriltag-detector, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\jazzy

        Environment: jazzy2
           Features: jazzy2, default
           Channels: D:/rj/output, https://fast.prefix.dev/conda-forge
   Dependency count: 12
       Dependencies: ros-jazzy-desktop-full, ros-jazzy-apriltag-msgs, ros-jazzy-ament-cmake, colcon-common-extensions, ros-jazzy-ament-cmake-clang-format, python, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\jazzy2

        Environment: jazzy3
           Features: jazzy3, default
           Channels: https://prefix.dev/robostack-jazzy, https://fast.prefix.dev/conda-forge
   Dependency count: 15
       Dependencies: ros-jazzy-desktop-full, python, ros-jazzy-image-view, ros-jazzy-apriltag, ros-jazzy-apriltag-msgs, ros-jazzy-ament-cmake, colcon-common-extensions, ros-jazzy-ament-cmake-clang-format, ros-jazzy-apriltag-detector, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\jazzy3
```

gftabor, 2025-06-13:
You can see the Cmake here that is causing it, the library and the executable are put at different levels of the directory. 
https://github.com/ros-misc-utilities/apriltag_detector/blob/master/apriltag_detector/CMakeLists.txt#L78
On Linux this is fine because the hook for ld_library_path adds the "$colcon_current_prefix/lib" to LD_LIBRARY_PATH

Tobias-Fischer, 2025-06-14:
Hopefully this is a pretty isolated problem - I remember at some stage @traversaro went through all packages and fixed up the install command, and also opened upstream PRs. So I suggest just fixing apriltag detector and we should be good :)

traversaro, 2025-06-14:
> > Hopefully this is a pretty isolated problem - I remember at some stage [@traversaro](https://github.com/traversaro) went through all packages and fixed up the install command, and also opened upstream PRs. So I suggest just fixing apriltag detector and we should be good :)
> 
> Ahh I see
> 
> [ros-jazzy/patch/ros-jazzy-rsl.win.patch](https://github.com/RoboStack/ros-jazzy/blob/1050b0546adcf7d35ab66d6da74827762cda0beb/patch/ros-jazzy-rsl.win.patch#L21)
> 
> Line 21 in [1050b05](/RoboStack/ros-jazzy/commit/1050b0546adcf7d35ab66d6da74827762cda0beb)
>  +    set(CMAKE_RUNTIME_OUTPUT_DIRECTORY "${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_BINDIR}")

No, that line is related to the location of build artifacts, not installed artifacts.

traversaro, 2025-06-14:
> Yes, the install commands in https://github.com/ros-misc-utilities/apriltag_detector/blob/master/apriltag_detector/CMakeLists.txt#L81-L85 are wrong for Windows. One can install the installation directories for all types, or simply remove the `DESTINATION lib`, as since the minimum cmake for the project is 3.16, that version already supports default destinations for `install(TARGETS` (see https://cmake.org/cmake/help/v3.16/command/install.html#installing-files), so not specifying any `DESTINATION` argument, all the artifacts will be installed in the right location.

I proposed a change in that direction in https://github.com/ros-misc-utilities/apriltag_detector/pull/7 .

gftabor, 2025-06-14:
> > Yes, the install commands in https://github.com/ros-misc-utilities/apriltag_detector/blob/master/apriltag_detector/CMakeLists.txt#L81-L85 are wrong for Windows. One can install the installation directories for all types, or simply remove the `DESTINATION lib`, as since the minimum cmake for the project is 3.16, that version already supports default destinations for `install(TARGETS` (see https://cmake.org/cmake/help/v3.16/command/install.html#installing-files), so not specifying any `DESTINATION` argument, all the artifacts will be installed in the right location.
> 
> I proposed a change in that direction in [ros-misc-utilities/apriltag_detector#7](https://github.com/ros-misc-utilities/apriltag_detector/pull/7) .

That all makes sense, their cmake had comments explaining the file structure was intentional for allowing the correct visibility in case other packages needed to link against the library. Is that not necessary? Not knowing what package to test against was why I was worried about changing the install directory.

closed 2025-06-16 by gftabor

Tobias-Fischer, 2025-06-25:
I think this should be all fine now? Feel free to reopen if not :)