RoboStack/ros-jazzy #74: Image view doesn't work

gftabor, 2025-06-16:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

```
(robostack:jazzy3) C:\Users\tabor\robostack>ros2 run image_view image_view
[INFO] [1750096155.809586600] [image_view_node]: Using transport "raw"
OpenCV: terminate handler is called! The last OpenCV error is:
OpenCV(4.11.0) Error: Unspecified error (The function is not implemented. Rebuild the library with Windows, GTK+ 2.x or Cocoa support. If you are on Ubuntu or Debian, install libgtk2.0-dev and pkg-config, then re-run cmake or configure script) in cvNamedWindow, file D:\bld\libopencv_1745417731047\work\modules\highgui\src\window.cpp, line 1284
[ros2run]: Process exited with failure 3221226505
```

Seems like

### Installed packages

```shell
https://gist.github.com/gftabor/e58954cb25e4474be71cf7a4e2e550a5

Pertinent 
py-opencv                                         4.11.0                 headless_py312h7c91e45_7  1.1 MiB     conda  https://fast.prefix.dev/conda-forge/
```

### Environment info

```shell
(robostack:jazzy3) C:\Users\tabor\robostack>pixi info
System
------------
       Pixi version: 0.48.0
           Platform: win-64
   Virtual packages: __win=10.0.26100=0
                   : __cuda=12.9=0
                   : __archspec=1=skylake
          Cache dir: C:\Users\tabor\AppData\Local\rattler\cache
       Auth storage: C:\Users\tabor\.rattler\credentials.json
   Config locations: No config files found

Global
------------
            Bin dir: C:\Users\tabor\.pixi\bin
    Environment dir: C:\Users\tabor\.pixi\envs
       Manifest dir: C:\Users\tabor\.pixi\manifests\pixi-global.toml

Workspace
------------
               Name: robostack
            Version: 0.1.0
      Manifest file: C:\Users\tabor\robostack\pixi.toml
       Last updated: 15-06-2025 15:23:10

Environments
------------
        Environment: default
           Features: default
           Channels: https://fast.prefix.dev/conda-forge
   Dependency count: 7
       Dependencies: python, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\default

        Environment: noetic
           Features: noetic, default
           Channels: https://prefix.dev/robostack-noetic, https://fast.prefix.dev/conda-forge
   Dependency count: 10
       Dependencies: ros-noetic-desktop, catkin_tools, rosdep, python, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\noetic

        Environment: humble
           Features: humble, default
           Channels: https://prefix.dev/robostack-humble, https://fast.prefix.dev/conda-forge
   Dependency count: 9
       Dependencies: ros-humble-desktop, rosdep, python, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\humble

        Environment: jazzy
           Features: jazzy, default
           Channels: https://prefix.dev/robostack-jazzy, https://fast.prefix.dev/conda-forge
   Dependency count: 9
       Dependencies: ros-jazzy-desktop-full, python, ros-jazzy-apriltag-detector, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\jazzy

        Environment: jazzy2
           Features: jazzy2, default
           Channels: D:/rj/output, https://fast.prefix.dev/conda-forge
   Dependency count: 12
       Dependencies: ros-jazzy-desktop-full, ros-jazzy-apriltag-msgs, ros-jazzy-ament-cmake, colcon-common-extensions, ros-jazzy-ament-cmake-clang-format, python, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\jazzy2

        Environment: jazzy3
           Features: jazzy3, default
           Channels: https://prefix.dev/robostack-jazzy, https://fast.prefix.dev/conda-forge
   Dependency count: 12
       Dependencies: ros-jazzy-desktop-full, python, ros-jazzy-ament-cmake, colcon-common-extensions, ros-jazzy-ament-cmake-clang-format, ros-jazzy-ament-cmake-vendor-package, compilers, cmake, pkg-config, make, ninja, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\jazzy3
```

traversaro, 2025-06-16:
> Can we somehow put that constraint in the mutex? Not sure if it’s possible with the new recipe format

I was thinking that having the ability of installing the headless version of opencv was still useful in some use cases, so I looked for all the packages that actually used `imshow` and added the dependency in it in https://github.com/RoboStack/ros-jazzy/pull/75 .

gftabor, 2025-06-16:
> libopencv should default to the `qt6` variant, but somehow sometimes the `headless` variant is installed, see for example [conda-forge/opencv-feedstock#472 (comment)](https://github.com/conda-forge/opencv-feedstock/issues/472#issuecomment-2937132243) . As a quick workaround, you can add a dependency on the specific qt6 variant of libopencv with:
> 
> ```
> libopencv = { version = "*", build = "*qt6*" }
> ```
> 
> the proper solution is to add this as a run dependency in the packages that need it.

Windows
https://gist.github.com/gftabor/fdabe8c993dbc40634300b8fe27ec5b3

Linux
https://gist.github.com/gftabor/9766ab524453b1f4fb58cb6c74d75ac7

Doesn't work for Linux or Windows as far as I can tell
```
[feature.jazzy.dependencies]
ros-jazzy-desktop-full = "*"
colcon-common-extensions = "*"
rosdep = "*"
python = "3.12.*"
libopencv = { version = "*", build = "*qt6*" }
```

traversaro, 2025-06-16:
There is some conflict of some kind. In these cases, I typically add more constraints of the solution I would like to get a more clear error.

traversaro, 2025-06-17:
Reopening as I suspect if there is a conflict of the type reported by  @gftabor   older package simply gets installed, so I want to still check something.

traversaro, 2025-06-17:
> Reopening as I suspect if there is a conflict of the type reported by [@gftabor](https://github.com/gftabor) older package simply gets installed, so I want to still check something.

I isolated the problem, and it is not robostack related:

~~~
traversaro@IITBMP014LW012:~$ conda create -n testjazzy -c conda-forge qt6-main==6.9.* py-opencv=4.11.*=*qt6*
Channels:
 - conda-forge
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: failed

LibMambaUnsatisfiableError: Encountered problems while solving:
  - package libopencv-4.11.0-qt6_py39h47cf047_600 requires qt6-main >=6.8.2,<6.9.0a0, but none of the providers can be installed

Could not solve for environment specs
The following packages are incompatible
├─ py-opencv =4.11 *qt6* is installable with the potential options
│  ├─ py-opencv 4.11.0 would require
│  │  └─ libopencv [==4.11.0 qt6_py310h32d48f6_603|==4.11.0 qt6_py310h4ebb18f_600|...|==4.11.0 qt6_py39hd96f159_602], which requires
│  │     └─ qt6-main >=6.8.2,<6.9.0a0 *, which can be installed;
│  └─ py-opencv 4.11.0 would require
│     └─ libopencv [==4.11.0 qt6_py310h070d3fe_604|==4.11.0 qt6_py310h070d3fe_605|...|==4.11.0 qt6_py39h2c622e1_607], which requires
│        └─ qt6-main >=6.8.3,<6.9.0a0 *, which can be installed;
└─ qt6-main =6.9 * is not installable because it conflicts with any installable versions previously reported.
~~~

traversaro, 2025-06-17:
> I isolated the problem, and it is not robostack related:
>
> ```
> traversaro@IITBMP014LW012:~$ conda create -n testjazzy -c conda-forge qt6-main==6.9.* py-opencv=4.11.*=*qt6*
> Channels:
>  - conda-forge [...]

I opened upstream issue: https://github.com/conda-forge/opencv-feedstock/issues/474 .

traversaro, 2025-06-17:
> What if we try to move to opencv 4.10 instead of 4.11?

This would mean doing a full rebuild with opencv 4.10, and possibily creating other incompatbilities with latest conda-forge packages as the latest pinning of conda-forge is built with opencv 4.11 .

traversaro, 2025-06-19:
@gftabor can you confirm this was solved by https://github.com/conda-forge/opencv-feedstock/pull/477 ? On other platforms we may need to wait for https://github.com/conda-forge/opencv-feedstock/pull/478, but for the specific problem on `win-64` and `linux-64`, I think this should be now solved.

traversaro, 2025-06-19:
Updated the title to capture the current state.

gftabor, 2025-06-19:
> [@gftabor](https://github.com/gftabor) can you confirm this was solved by [conda-forge/opencv-feedstock#477](https://github.com/conda-forge/opencv-feedstock/pull/477) ? On other platforms we may need to wait for [conda-forge/opencv-feedstock#478](https://github.com/conda-forge/opencv-feedstock/pull/478), but for the specific problem on `win-64` and `linux-64`, I think this should be now solved.

Looks like I have install conflicts on Linux still.

gftabor, 2025-06-19:
> Exactly the same error or different?

It seems maybe prefix.dev conda list has not updated yet with new file on linux. When I switch channel from prefix.dev/conda-forge to just conda-forge it can install.

closed 2025-07-26 by Tobias-Fischer