RoboStack/ros-jazzy #78: Package request: ros-jazzy-desktop (unique version number for py311)

deltaniel, 2025-06-19:
### Package name

ros-jazzy-desktop

### Operating System(s)

- [x] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [ ] osx-arm64
- [x] win-64

### Additional context (optional)

I have been using robostack-jazzy with pixi and python 3.11 for win-64 and linux-64. In the recent python version bump to 3.12, several packages were published with the same version number, causing dependencies to break if the pixi lock file is regenerated. One solution is to specify build strings, but this can be impractical when working across multiple platforms.

Is it possible for the version numbers to be updated to distinguish between python 3.11 and 3.12? I have noticed this issue for the following packages, although there are likely more:

- [ros-jazzy-desktop](https://prefix.dev/channels/robostack-jazzy/packages/ros-jazzy-desktop)
- [ros-jazzy-joint-state-publisher-gui](https://prefix.dev/channels/robostack-jazzy/packages/ros-jazzy-joint-state-publisher-gui)
- [ros-jazzy-ament-cmake-auto](https://prefix.dev/channels/robostack-jazzy/packages/ros-jazzy-ament-cmake-auto)

Thanks! (Super appreciate how RoboStack makes ROS so accessible)

Tobias-Fischer, 2025-06-19:
Hi @deltaniel - many thanks for your kind words!

Would it be possible for you to provide us with a minimal reproducible example – including where it breaks and what the expected behaviour would be? That would enable us to pinpoint down what are wrong a lot better :)

deltaniel, 2025-06-20:
Of course! 
### Minimum reproducible example
Set up a new [pixi](https://pixi.sh/latest/) project with `pixi.toml`:

```toml
[project]
name = "test"
channels = ["conda-forge", "robostack-jazzy"]
platforms = ["win-64", "linux-64"]

[dependencies]
python = "3.11.*"
ros-jazzy-desktop = "*"

```
Breaks when running `pixi install`:
<pre><details> <summary> Full error message </summary>
Error:   × failed to solve the conda requirements of 'default' 'win-64'
  ╰─▶ Cannot solve the request because of: The following packages are incompatible                                                                                                                                                                                                                                     
      ├─ python 3.11.* can be installed with any of the following options:
      │  └─ python 3.11.0 | 3.11.0 | 3.11.1 | 3.11.2 | 3.11.3 | 3.11.4 | 3.11.5 | 3.11.6 | 3.11.7 | 3.11.7 | 3.11.8 | 3.11.9 | 3.11.10 | 3.11.10 | 3.11.10 | 3.11.10 | 3.11.11 | 3.11.11 | 3.11.12 | 3.11.13
      └─ ros-jazzy-desktop * cannot be installed because there are no viable options:
         ├─ ros-jazzy-desktop 0.11.0 | 0.11.0 would require
         │  └─ python_abi 3.12.* *_cp312, which cannot be installed because there are no viable options:
         │     └─ python_abi 3.12 | 3.12 | 3.12 | 3.12 would constrain
         │        └─ python 3.12.* *_cpython, which conflicts with any installable versions previously reported
         └─ ros-jazzy-desktop 0.11.0 | 0.11.0 would require
            └─ ros-jazzy-turtlesim *, which cannot be installed because there are no viable options:
               ├─ ros-jazzy-turtlesim 1.8.3 | 1.8.3 would require
               │  └─ python_abi 3.12.* *_cp312, which cannot be installed because there are no viable options:
               └─ ros-jazzy-turtlesim 1.8.3 | 1.8.3 would require
                  └─ ros-jazzy-rclcpp-action *, which cannot be installed because there are no viable options:
                     ├─ ros-jazzy-rclcpp-action 28.1.8 | 28.1.9 would require
                     │  └─ python_abi 3.12.* *_cp312, which cannot be installed because there are no viable options:
                     └─ ros-jazzy-rclcpp-action 28.1.6 | 28.1.6 would require
                        └─ ros-jazzy-ament-cmake *, which cannot be installed because there are no viable options:
                           ├─ ros-jazzy-ament-cmake 2.5.3 | 2.5.4 would require
                           │  └─ python_abi 3.12.* *_cp312, which cannot be installed because there are no viable options:
                           └─ ros-jazzy-ament-cmake 2.5.3 | 2.5.3 would require
                              └─ ros-jazzy-ament-cmake-core *, which cannot be installed because there are no viable options:
                                 └─ ros-jazzy-ament-cmake-core 2.5.4 would require
                                    └─ python_abi 3.12.* *_cp312, which cannot be installed because there are no viable options:
</details></pre>

### Expected behavior
Pixi should install [`ros-jazzy-desktop`](https://prefix.dev/channels/robostack-jazzy/packages/ros-jazzy-desktop) v0.11.0 with build string `np126py311h7d8ee0e_4` (for win-64) or `np126py311hdfa6bdf_4` (for linux-64), which are compatible with python 3.11.

Tobias-Fischer, 2025-06-20:
Very interesting, thanks! Seems like some issue in pixi - ccing @ruben-arts @wolfv to please take a look :)

traversaro, 2025-06-20:
> Pixi should install [`ros-jazzy-desktop`](https://prefix.dev/channels/robostack-jazzy/packages/ros-jazzy-desktop) v0.11.0 with build string `np126py311h7d8ee0e_4` (for win-64)

I am wrong, or it seems that the issue is that this package is not installable at all at the moment on Windows? See

~~~
C:\Users\straversaro>micromamba create -n test78 -c conda-forge -c robostack-jazzy ros-jazzy-desktop="0.11.*=*_4"
conda-forge/noarch                                          Using cache
conda-forge/win-64                                          Using cache
robostack-jazzy/noarch                                      Using cache
robostack-jazzy/win-64                                      Using cache

error    libmamba Could not solve for environment specs
    The following package could not be installed
    └─ ros-jazzy-desktop =0.11 *_4 is not installable because it requires
       ├─ ros-jazzy-angles =* * with the potential options
       │  ├─ ros-jazzy-angles 1.16.0 would require
       │  │  └─ ros-jazzy-ament-cmake =* * with the potential options
       │  │     ├─ ros-jazzy-ament-cmake 2.5.3 would require
       │  │     │  └─ ros2-distro-mutex =0.6 jazzy_*, which can be installed;
       │  │     ├─ ros-jazzy-ament-cmake [2.5.3|2.5.4] would require
       │  │     │  ├─ ros-jazzy-ament-cmake-core =* *, which requires
       │  │     │  │  └─ ros2-distro-mutex =0.9 jazzy_*, which can be installed;
       │  │     │  └─ ros2-distro-mutex =0.9 jazzy_*, which can be installed;
       │  │     └─ ros-jazzy-ament-cmake 2.5.3 would require
       │  │        └─ ros2-distro-mutex =0.8 jazzy_*, which can be installed;
       │  ├─ ros-jazzy-angles 1.16.0 would require
       │  │  └─ ros2-distro-mutex =0.6 jazzy_*, which can be installed;
       │  ├─ ros-jazzy-angles 1.16.0 would require
       │  │  └─ ros2-distro-mutex =0.8 jazzy_*, which can be installed;
       │  └─ ros-jazzy-angles 1.16.0 would require
       │     └─ ros2-distro-mutex =0.9 jazzy_*, which can be installed;
       └─ ros2-distro-mutex =0.7 jazzy_*, which conflicts with any installable versions previously reported.
critical libmamba Could not solve for environment specs
~~~

It is as if a build of ros-jazzy-angles is missing.

deltaniel, 2025-06-20:
> I am wrong, or it seems that the issue is that this package is not installable at all at the moment on Windows?

I've tried iteratively pinning specific builds, and it looks like the necessary build for `ros-jazzy-ament-cmake-core` is missing:
```diff
# pixi.toml
[project]
name = "test"
channels = ["conda-forge", "robostack-jazzy"]
- platforms = ["win-64", "linux-64"]
+ platforms = ["win-64"]

[dependencies]
python = "3.11.*"
- ros-jazzy-desktop = "*"
+ ros-jazzy-desktop = { version = "==0.11.0", build = "np126py311h7d8ee0e_4" }
+ ros-jazzy-turtlesim = { version = "==1.8.3", build = "np126py311h7d8ee0e_4"}
+ ros-jazzy-ament-cmake = { version = "==2.5.3", build = "np126py311h7d8ee0e_4"}
+ ros-jazzy-ament-cmake-core = { version = "==2.5.3", build = "np126py311h7d8ee0e_4"}
```

<pre><details><summary>pixi install</summary>
Error:                                                                                                                                                                                                                                  
  × failed to solve the conda requirements of 'default' 'win-64'                                                                                                                                                                        
  ╰─▶ Cannot solve the request because of: No candidates were found for ros-jazzy-ament-cmake-core ==2.5.3 np126py311h7d8ee0e_4.
</details></pre>

`ros-jazzy-ament-cmake-core 2.5.3=np126py311h7d8ee0e_4` seems to be missing, despite it being listed on [prefix.dev](https://prefix.dev/channels/robostack-jazzy/packages/ros-jazzy-ament-cmake-core) (notice that the output does not list `(+ _ builds)` at the start or `Other Versions` at the end):

<pre><details><summary>pixi search ros-jazzy-ament-cmake-core --channel robostack-jazzy</summary>
Using channels: robostack-jazzy

ros-jazzy-ament-cmake-core-2.5.4-np126py312h4d29ed4_7
-----------------------------------------------------

Name                ros-jazzy-ament-cmake-core
Version             2.5.4
Build               np126py312h4d29ed4_7
Size                40950
License             BSD-3-Clause
Subdir              win-64
File Name           ros-jazzy-ament-cmake-core-2.5.4-np126py312h4d29ed4_7.conda
URL                 https://conda.anaconda.org/robostack-jazzy/win-64/ros-jazzy-ament-cmake-core-2.5.4-np126py312h4d29ed4_7.conda
MD5                 ecadc50a5f4e09048f90b2e2d5552742
SHA256              9a124e5bfd2657b7a991aec31c20768bd806afaaae69bc0b070e651d2ccb06c2

Dependencies:
 - catkin_pkg
 - python
 - ros-jazzy-ament-package
 - ros2-distro-mutex 0.9.* jazzy_*
 - cmake
 - vc >=14.3,<15
 - vc14_runtime >=14.42.34438
 - ucrt >=10.0.20348.0
 - vc >=14.3,<15
 - vc14_runtime >=14.42.34438
 - ucrt >=10.0.20348.0
 - ros2-distro-mutex >=0.9.0,<0.10.0a0
 - numpy >=1.26.4,<2.0a0
 - python_abi 3.12.* *_cp312
</details></pre>

traversaro, 2025-06-25:
> Also just to clarify something for myself, was my original assumption in the issue incorrect in that it doesn't matter if there are different builds for python 3.11 and 3.12 that share the same version number?

Yes, as long as the hash (i.e. build string) or the build number are different, it is perfectly fine.

closed 2025-07-26 by Tobias-Fischer