RoboStack/ros-jazzy #81: Windows   ros2 launch turtlebot3_navigation2 navigation2.launch.py crashes

gftabor, 2025-06-23:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Some of the nav stack nodes crash launching turtlebot3 navigation launch. I am testing in simulation if that matters.

terminal 1
```
set TURTLEBOT3_MODEL=burger
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
```

Terminal 2

```
set TURTLEBOT3_MODEL=burger
ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True
```

Results in 

```
(robostack:jazzy) C:\Users\tabor\robostack>ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True
[INFO] [launch]: All log files can be found below C:\Users\tabor\.ros\log\2025-06-23-17-50-13-434109-tabor-16768
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_isolated.EXE-1]: process started with pid [31444]
[INFO] [rviz2.EXE-2]: process started with pid [45176]
[component_container_isolated.EXE-1] [INFO] [1750715414.603098300] [nav2_container]: Load Library: C:\Users\tabor\robostack\.pixi\envs\jazzy\Library/bin/map_server_core.dll
[component_container_isolated.EXE-1] [INFO] [1750715414.617476400] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::CostmapFilterInfoServer>
[component_container_isolated.EXE-1] [INFO] [1750715414.617513000] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapSaver>
[component_container_isolated.EXE-1] [INFO] [1750715414.617516400] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapServer>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/map_server' in container '/nav2_container'
[component_container_isolated.EXE-1] [INFO] [1750715414.617530100] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapServer>
[component_container_isolated.EXE-1] [INFO] [1750715414.627170800] [map_server]:
[component_container_isolated.EXE-1]    map_server lifecycle node launched.
[component_container_isolated.EXE-1]    Waiting on external lifecycle transitions to activate
[component_container_isolated.EXE-1]    See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated.EXE-1] [INFO] [1750715414.627200100] [map_server]: Creating
[component_container_isolated.EXE-1] [INFO] [1750715414.627652500] [nav2_container]: Load Library: C:\Users\tabor\robostack\.pixi\envs\jazzy\Library/bin/amcl_core.dll
[component_container_isolated.EXE-1] [INFO] [1750715414.632972800] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_amcl::AmclNode>
[component_container_isolated.EXE-1] [INFO] [1750715414.633010100] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_amcl::AmclNode>
[component_container_isolated.EXE-1] [INFO] [1750715414.637996300] [amcl]:
[component_container_isolated.EXE-1]    amcl lifecycle node launched.
[component_container_isolated.EXE-1]    Waiting on external lifecycle transitions to activate
[component_container_isolated.EXE-1]    See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/amcl' in container '/nav2_container'
[component_container_isolated.EXE-1] [INFO] [1750715414.638620300] [amcl]: Creating
[component_container_isolated.EXE-1] [INFO] [1750715414.641058400] [nav2_container]: Load Library: C:\Users\tabor\robostack\.pixi\envs\jazzy\Library/bin/nav2_lifecycle_manager_core.dll
[component_container_isolated.EXE-1] [INFO] [1750715414.645460100] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_lifecycle_manager::LifecycleManager>
[component_container_isolated.EXE-1] [INFO] [1750715414.645496000] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_lifecycle_manager::LifecycleManager>
[rviz2.EXE-2] [INFO] [1750715415.336144000] [rviz2]: Stereo is NOT SUPPORTED
[rviz2.EXE-2] [INFO] [1750715415.336326500] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2.EXE-2] [INFO] [1750715415.370494600] [rviz2]: Stereo is NOT SUPPORTED
[rviz2.EXE-2] [INFO] [1750715415.429828600] [rviz2]: Subscribing to: /mobile_base/sensors/bumper_pointcloud
[rviz2.EXE-2]
[rviz2.EXE-2] [INFO] [1750715415.430319100] [rviz2]: Subscribing to: /mobile_base/sensors/bumper_pointcloud
[rviz2.EXE-2]
[rviz2.EXE-2] [INFO] [1750715415.440123600] [rviz2]: Subscribing to: /local_costmap/voxel_marked_cloud
[rviz2.EXE-2]
[rviz2.EXE-2] [INFO] [1750715415.496963100] [rviz2]: Subscribing to: /global_costmap/voxel_marked_cloud
[rviz2.EXE-2]
[rviz2.EXE-2] [INFO] [1750715415.837421000] [rviz2]: Subscribing to: /mobile_base/sensors/bumper_pointcloud
[rviz2.EXE-2]
[rviz2.EXE-2] [INFO] [1750715415.869431600] [rviz2]: Subscribing to: /local_costmap/voxel_marked_cloud
[rviz2.EXE-2]
[rviz2.EXE-2] [INFO] [1750715415.932611200] [rviz2]: Subscribing to: /global_costmap/voxel_marked_cloud
[rviz2.EXE-2]
[rviz2.EXE-2] [WARN] [1750715416.045457500] [rcl.logging_rosout]: Publisher already registered for node name: 'rviz2'. If this is due to multiple nodes with the same name then all logs for the logger named 'rviz2' will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[... 65 lines cut ...]

        Environment: noetic
           Features: noetic, default
           Channels: https://prefix.dev/robostack-noetic, https://prefix.dev/conda-forge
   Dependency count: 12
       Dependencies: ros-noetic-desktop, catkin_tools, rosdep, python, compilers, cmake, pkg-config, make, ninja, pip, apriltag, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\noetic

        Environment: humble
           Features: humble, default
           Channels: https://prefix.dev/robostack-humble, https://prefix.dev/conda-forge
   Dependency count: 11
       Dependencies: ros-humble-desktop, rosdep, python, compilers, cmake, pkg-config, make, ninja, pip, apriltag, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\humble

        Environment: jazzy
           Features: jazzy, default
           Channels: https://prefix.dev/robostack-jazzy, https://prefix.dev/conda-forge
   Dependency count: 21
       Dependencies: ros-jazzy-desktop-full, python, ros-jazzy-turtlebot3, ros-jazzy-turtlebot3-msgs, ros-jazzy-navigation2, ros-jazzy-nav2-bringup, ros-jazzy-turtlebot3-simulations, colcon-common-extensions, rosdep, ros-jazzy-ament-cmake-vendor-package, ros-jazzy-apriltag-msgs, ros-jazzy-ament-cmake-clang-format, ros-jazzy-apriltag, compilers, cmake, pkg-config, make, ninja, pip, apriltag, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\jazzy
```

traversaro, 2025-06-23:
Is this with conda-forge's ogre installed or not? I suspect there may be more conflicts there in vendored ogre libraries.

gftabor, 2025-06-24:
> Is this with conda-forge's ogre installed or not? I suspect there may be more conflicts there in vendored ogre libraries.

Desktop-full so definitely has conda-forge ogre installed. 

I suspected the rviz crashes at the end were related to 
#57 , but the nav stack component container crashing as well was the main thing I noticed.

traversaro, 2025-06-25:
Thanks for the investigation. `3221226505` means `0xC0000409` that means `STATUS_STACK_BUFFER_OVERRUN`. Probably having some sort of backtrace would be the wayto investigate. Probably running it directly from the `lifecycle_manager` executable instead of via `ros2 run` could also provide more informative errors I guess.

gftabor, 2025-06-25:
Ahh there is a patch for this in robostack, missed it.

gftabor, 2025-07-01:
I have been able to track the crashing to this exact line.

https://github.com/ros-navigation/navigation2/blob/jazzy/nav2_lifecycle_manager/src/lifecycle_manager.cpp#L109

Commenting it out gets the full nav stuck to run.

gftabor, 2025-07-07:
Building these packages
https://github.com/RoboStack/ros-jazzy/pull/90
 with these patches from source with colcon did solve my problem locally. However, it doesn't seem like the new conda packages act the same. Still more digging before I can close this issue.

On a related note, I was hoping to merge into nav2 all the patches we have. 

https://github.com/ros-navigation/navigation2/pull/5333

Not totally sure if @traversaro had a reason behind not trying to merge them. There definitely are a few where balancing windows building and linux code cleanliness might be difficult.

traversaro, 2025-07-08:
> 
> > On a related note, I was hoping to merge into nav2 all the patches we have.
> > [ros-navigation/navigation2#5333](https://github.com/ros-navigation/navigation2/pull/5333)
> > Not totally sure if [@traversaro](https://github.com/traversaro) had a reason behind not trying to merge them. There definitely are a few where balancing windows building and linux code cleanliness might be difficult.
> 
> Thanks! To be honest I do not even remember if I authored those patches or if they carried over from different distros. Anyhow, now specific reason why I did not tried to upstream them before, other then that we have a huge upstreaming backlog (see [RoboStack/robostack.github.io#16](https://github.com/RoboStack/robostack.github.io/issues/16)) and I was going through other things now, so thanks a lot for opening the PR. Probably some of the reason why I did not started the upstreaming with the nav2 patches is that there were a few things that had to be cleaned up, but I think that as long as we address Steve comments, we are good to go.

Actually after looking a bit into this, I think what I did was to start integrating the patches in the main branches, but then I never looked in how to backport those changes.

traversaro, 2025-07-09:
> Obviously we don't want to do a full rebuild, so now I am going to try to figure out which packages actually need to be rebuilt. Probably just all nav2?

Actually we need to do a full rebuild anyhow for `rmw_zenoh_cpp` compatibility, see https://discourse.ros.org/t/wire-breaking-change-in-rmw-zenoh-for-ros-2-jazzy and https://discourse.ros.org/t/new-packages-for-jazzy-jalisco-2024-07-29 . Note that the title of the second link is wrong, but the breaking `rmw_zenoh_cpp` 0.2.5 release went out on the 29th of June, so since then the default install or `rmw_zenoh_cpp` via robostack/conda/pixi and via apt are not wire compatible, so I think the simplest thing to do is just to do a rebuild (I want also to port a few improvements to the rebuild infrastructure done in https://github.com/RoboStack/ros-humble/pull/320).

> Not totally sure what to do on the pointer one.

Which one is the pointer one?

closed 2025-07-26 by Tobias-Fischer