RoboStack/ros-jazzy #87: Package request: ros-jazzy-ament-cmake-core-2.5.3-np126py311hdfa6bdf_4

TomNysWF, 2025-06-25:
### Package name

ros-jazzy-ament-cmake-core

### Operating System(s)

- [x] linux-64
- [x] linux-aarch64
- [x] osx-64
- [x] osx-arm64
- [x] win-64

### Additional context (optional)

I am using Pixi to manage packages and am trying to get my packages but i am getting the following error, i can see the package is there in prefix.dev, but not on the anaconda site it is refering to.

`error:
  × failed to fetch ros-jazzy-ament-cmake-core-2.5.3-np126py311hdfa6bdf_4.conda
  ╰─▶ HTTP status client error (404 Not Found) for url (https://conda.anaconda.org/robostack-jazzy/linux-64/ros-jazzy-ament-cmake-core-2.5.3-np126py311hdfa6bdf_4.conda)`

When using the channels = ["https://prefix.dev/robostack-jazzy"] it does seems to find the package. But by default we use the "robostack-jazzy" channel

traversaro, 2025-06-25:
Can you shared the exact `pixi.toml` you are using? Thanks!

TomNysWF, 2025-06-25:
@traversaro I added the files above

traversaro, 2025-06-25:
> So, the file:
> 
>     * https://repo.prefix.dev/robostack-jazzy/linux-64/ros-jazzy-ament-cmake-version-2.5.3-np126py311hdfa6bdf_4.conda
> 
> 
> can be downloaded, while https://conda.anaconda.org/robostack-jazzy/linux-64/ros-jazzy-ament-cmake-core-2.5.3-np126py311hdfa6bdf_4.conda no. Interestingly, if I look in the website the file to be downloaded is https://anaconda.org/robostack-jazzy/ros-jazzy-ament-cmake-version/2.5.3/download/linux-64/ros-jazzy-ament-cmake-version-2.5.3-np126py311hdfa6bdf_4.conda (without the `conda.` prefix in the url) and it works fine.

Not sure  why I looked for `ros-jazzy-ament-cmake-version`, the problem seems ros-jazzy-ament-cmake-core, and somehow it seems that the first upload for it was just ~13 days ago.

traversaro, 2025-06-25:
Thanks for reporting the problem. I reported the issue upstream to anaconda.org at https://github.com/conda/infrastructure/issues/1171 . We do not manage directly the anaconda.org server, so until the issue is solved, please use https://prefix.dev/robostack-jazzy as a channel until the problem is solved.

traversaro, 2025-06-25:
As a workaround we can setup a bot that mirrors back packages from prefix.dev robostack-jazzy to anaconda.org's robostack-jazzy (for example adapting https://github.com/RoboStack/robostack.github.io/blob/master/copy-to-distro-specific-channel.py), but for a few days I would like to leave the situation as it is to permit to anaconda.org staff to investigate what happened.

traversaro, 2025-06-26:
Ok, now the packages should be available.

TomNysWF, 2025-06-26:
Perfect it seems to be back, thanks 🥳

closed 2025-06-26 by TomNysWF