RoboStack/ros-jazzy #93: On-wire incompatibility when using rmw_zenoh_cpp between RoboStack ROS Jazzy default install and apt ROS Jazzy default install since 2025-06-29

traversaro, 2025-07-09:
On 2025-06-29 the rmw_zenoh_cpp  on the main apt/deb official channel of ROS Jazzy was updated to 0.2.5, that is not anymore on wire compatible with earlier versions (i.e. <= 0.2.3). 

In a nutshell, this means that if a users install ROS Jazzy with .deb packages on a machine, and ROS Jazzy with RoboStack on another machine, if it tries to communicate, the node will crash with errors like:

~~~
[robot_state_publisher-5] thread 'rx-7' panicked at library/core/src/panicking.rs:136:5:
[robot_state_publisher-5] panic in a function that cannot unwind
[robot_state_publisher-5] note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[robot_state_publisher-5] thread caused non-unwinding panic. aborting.
[rviz2-3] thread 'rx-5' panicked at library/core/src/panicking.rs:136:5:
[rviz2-3] panic in a function that cannot unwind
[rviz2-3] note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[rviz2-3] thread caused non-unwinding panic. aborting.
[gzserver-1] thread 'rx-1' panicked at library/core/src/panicking.rs:136:5:
[gzserver-1] panic in a function that cannot unwind
[gzserver-1] note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[gzserver-1] thread caused non-unwinding panic. aborting.
~~~

The best solution is to update also RoboStack's rmw_zenoh_cpp with a good old full rebuild/sync.

gftabor, 2025-07-11:
I am wanting to try out zenoh whats the bottleneck on triggering the rebuilt?

Tobias-Fischer, 2025-07-11:
I think we just need to bump the build / mutex version number, open a PR, and see what happens ;)

Ideally we should bring across any infrastructure fixes from the currently ongoing humble rebuild (which should fix https://github.com/RoboStack/ros-jazzy/issues/86) and also tackle https://github.com/RoboStack/vinca/issues/88 while we are at it.

traversaro, 2025-07-11:
I have part of this locally, I plan to push it tomorrow!

traversaro, 2025-07-15:
> I have part of this locally, I plan to push it tomorrow!

Tomorrow became ~4 days later, but here we are: https://github.com/RoboStack/ros-jazzy/pull/95 .