RoboStack/ros-jazzy #94: nav2_smac_planner needs ODE added on windows

gftabor, 2025-07-09:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Following up here
https://github.com/ros-navigation/navigation2/pull/5333#discussion_r2196144535

Might be helpful to use my PR copy
https://github.com/gftabor/navigation2

```
colcon build --merge-install --event-handlers desktop_notification- --event-handlers console_cohesion+  --cmake-args -DBUILD_TESTING=OFF --packages-select nav2_smac_planner
```

results in 

```
CMake Error at CMakeLists.txt:79 (target_link_libraries):
  Target "nav2_smac_planner" links to:

    ODE::ODE

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.
 ```

### Installed packages

```shell
https://gist.github.com/gftabor/2db97ae7661de04a9da73a5835b57ca8
```

### Environment info

```shell
(robostack:jazzy) C:\Users\tabor\robostack\nav_ws>pixi info
System
------------
       Pixi version: 0.48.0
           Platform: win-64
   Virtual packages: __win=10.0.26100=0
                   : __cuda=12.9=0
                   : __archspec=1=skylake
          Cache dir: C:\Users\tabor\AppData\Local\rattler\cache
       Auth storage: C:\Users\tabor\.rattler\credentials.json
   Config locations: No config files found

Global
------------
            Bin dir: C:\Users\tabor\.pixi\bin
    Environment dir: C:\Users\tabor\.pixi\envs
       Manifest dir: C:\Users\tabor\.pixi\manifests\pixi-global.toml

Workspace
------------
               Name: robostack
            Version: 0.1.0
      Manifest file: C:\Users\tabor\robostack\pixi.toml
       Last updated: 09-07-2025 13:06:25

Environments
------------
        Environment: default
           Features: default
           Channels: https://prefix.dev/conda-forge
   Dependency count: 9
       Dependencies: python, compilers, cmake, pkg-config, make, ninja, pip, apriltag, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\default

        Environment: noetic
           Features: noetic, default
           Channels: https://prefix.dev/robostack-noetic, https://prefix.dev/conda-forge
   Dependency count: 12
       Dependencies: ros-noetic-desktop, catkin_tools, rosdep, python, compilers, cmake, pkg-config, make, ninja, pip, apriltag, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\noetic

        Environment: humble
           Features: humble, default
           Channels: https://prefix.dev/robostack-humble, https://prefix.dev/conda-forge
   Dependency count: 11
       Dependencies: ros-humble-desktop, rosdep, python, compilers, cmake, pkg-config, make, ninja, pip, apriltag, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\humble

        Environment: jazzy
           Features: jazzy, default
           Channels: https://prefix.dev/robostack-jazzy, C:/Users/tabor/ros-jazzy/output, https://prefix.dev/conda-forge
   Dependency count: 24
       Dependencies: ros-jazzy-desktop, python, ros-jazzy-turtlebot3, ros-jazzy-turtlebot3-msgs, ros-jazzy-turtlebot3-simulations, colcon-common-extensions, rosdep, ros-jazzy-diagnostic-updater, ros-jazzy-nav2-lifecycle-manager, ros-jazzy-ament-cmake-google-benchmark, ros-jazzy-performance-test-fixture, ros-jazzy-mimick-vendor, ros-jazzy-ament-cmake-vendor-package, ros-jazzy-ament-cmake-clang-format, ros-jazzy-test-msgs, xtensor, compilers, cmake, pkg-config, make, ninja, pip, apriltag, vs2022_win-64
   Target platforms: win-64
    Prefix location: C:\Users\tabor\robostack\.pixi\envs\jazzy
```

traversaro, 2025-07-10:
Your code still has the patch leftover: 

~~~
list(REMOVE_ITEM OMPL_LIBRARIES "ode_double")
list(APPEND OMPL_LIBRARIES "ODE::ODE")
~~~ 

Can you remove it and retry?

gftabor, 2025-07-11:
You're right, totally builds. Slipped by me

closed 2025-07-11 by gftabor