RoboStack/ros-jazzy #103: Unable to colcon build packages

jfrancis71, 2025-09-10:
### Solution to issue cannot be found in the documentation.

- [x] I checked the documentation.

### Issue

Attempting to build packages with colcon build --symlink-install results in error:
```
--- stderr: test_1
/home/julian/miniforge3/envs/ros_jazzy/lib/python3.12/site-packages/setuptools/_distutils/dist.py:289: UserWarning: Unknown distribution option: 'tests_require'
  warnings.warn(msg)
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: option --editable not recognized
```

This seems to be the case for any package including a black package (as above) created by:
``ros2 pkg create --build-type ament_python test_1```

Tested in a newly created (as of 09/09/2025) mamba RoboStack Jazzy environment.

I notice that my previous RoboStack Jazzy environment used Python 3.11.11. This new RoboStack environment is using Python 3.12.11

I believe this is relevant: https://github.com/ros2/ros2/issues/1702#issuecomment-3081905180
See Amarillo-civ comment on July 17.

If I follow that suggestion, I can now build packages. I do not know if that is the correct or best solution, but it does seem to work in my case.

### Installed packages

```shell
# packages in environment at /home/julian/miniforge3/envs/ros_jazzy:
#
# Name                    Version                   Build  Channel
_openmp_mutex             4.5                       2_gnu    conda-forge
adwaita-icon-theme        48.1                     unix_1    conda-forge
aiohappyeyeballs          2.6.1              pyhd8ed1ab_0    conda-forge
aiohttp                   3.12.15         py312ha4530ae_0    conda-forge
aiosignal                 1.4.0              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.14               h86ecc28_0    conda-forge
aom                       3.9.1                hcccb83c_0    conda-forge
argcomplete               3.6.2              pyhd8ed1ab_0    conda-forge
assimp                    5.4.3                hdc325bc_0    conda-forge
at-spi2-atk               2.38.0               h1f2db35_3    conda-forge
at-spi2-core              2.40.3               h1f2db35_0    conda-forge
atk-1.0                   2.38.0               hedc4a1f_2    conda-forge
attr                      2.5.1                h4e544f5_1    conda-forge
attrs                     25.3.0             pyh71513ae_0    conda-forge
binutils                  2.44                 hf1166c9_1    conda-forge
binutils_impl_linux-aarch64 2.44                 h4c662bb_1    conda-forge
binutils_linux-aarch64    2.44                 hf1166c9_1    conda-forge
blosc                     1.21.6               hb4dfabd_1    conda-forge
brotli                    1.1.0                he30d5cf_4    conda-forge
brotli-bin                1.1.0                he30d5cf_4    conda-forge
bullet                    3.25                 h55cb1a1_4    conda-forge
bullet-cpp                3.25            py312ha2895bd_4    conda-forge
bzip2                     1.0.8                h68df207_7    conda-forge
c-ares                    1.34.5               h86ecc28_0    conda-forge
c-compiler                1.11.0               hdceaead_0    conda-forge
ca-certificates           2025.8.3             hbd8a1cb_0    conda-forge
cairo                     1.18.4               h83712da_0    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_1    conda-forge
catkin_tools              0.9.5              pyh29332c3_0    conda-forge
cffi                      1.17.1          py312h2fc7fbd_1    conda-forge
cli11                     2.5.0                h5ad3122_0    conda-forge
cmake                     4.1.1                hc9d863e_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_1    conda-forge
[... 702 lines cut ...]
    conda-build version : not installed
         python version : 3.12.8.final.0
                 solver : libmamba (default)
       virtual packages : __archspec=1=aarch64
                          __conda=24.11.2=0
                          __glibc=2.35=0
                          __linux=5.15.0=0
                          __unix=0=0
       base environment : /home/julian/miniforge3  (writable)
      conda av data dir : /home/julian/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-jazzy/linux-aarch64
                          https://conda.anaconda.org/robostack-jazzy/noarch
                          https://conda.anaconda.org/conda-forge/linux-aarch64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/julian/miniforge3/pkgs
                          /home/julian/.conda/pkgs
       envs directories : /home/julian/miniforge3/envs
                          /home/julian/.conda/envs
               platform : linux-aarch64
             user-agent : conda/24.11.2 requests/2.32.3 CPython/3.12.8 Linux/5.15.0-1085-raspi ubuntu/22.04.5 glibc/2.35 solver/libmamba conda-libmamba-solver/24.9.0 libmambapy/1.5.12
                UID:GID : 1000:1003
             netrc file : None
           offline mode : False
```

jfrancis71, 2025-09-15:
Presently, I've just down graded my setup tools package version which seems to work. I suspect other people might run into this problem. I'll hold this open until the end of the week to see if any other comments, but otherwise close it off.

traversaro, 2025-09-15:
xref: https://github.com/colcon/colcon-core/issues/696

Tobias-Fischer, 2025-11-04:
Some magic now pins setuptools<80 by default, so I think we can close.

closed 2025-11-04 by Tobias-Fischer